EdgeTX/edgetx #3680: CPN W64: simulator crashes on selection of SBUS or FlySky external module

mha1, 2023-06-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

W64 simulator crashes if SBUS or FlySky is chosen as external RF module.

### Expected Behavior

Don't crash

### Steps To Reproduce

1. standalone simulator
- start Firmware Simulator 2.9
- goto model settings and select external RF module
- select SBUS or FlySky

2. from Companion
- start Companion 2.9
- select File/New then click on +Model
- press cancel to exit the wizard
- double click on the newly created MODEL01
- select SBUS or AFHDS3 as external radio module
- press simualte

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### OS Version

Windows 10

### Anything else?

Companion W64 build:  Actions, main branch, Windows Companion 64-bit, chore: Translation catchup (#3619) - https://github.com/EdgeTX/edgetx/suites/13479597982/artifacts/740096546

Note: 

selecting external module SBUS doesn't crash a build with commit df00d7d8 but the following commit d80adcbc (#3055) does. Flysky does work with commit d80adcbc (#3055). This might indicate the crashes on SBUS and FlySky have independent root causes.

pfeerick, 2023-06-15:
Ah, #3055... the PR that just keeps on giving 😆

closed 2023-06-15 by pfeerick

mha1, 2023-06-15:
pure bliss

PR #3683 (2023-06-15): fix CPN: simulator crashes on selection of SBUS external module
Fixes #3680

Summary of changes:
- checks function ptr for nullptr before accessing it (simu doesn't implement function)