EdgeTX/edgetx #3682: CPN W64: simulator crashes on selection of FLYSKY external module

mha1, 2023-06-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

W64 simulator crashes if FLYSKY is chosen as external RF module.

### Expected Behavior

Don't crash

### Steps To Reproduce

1. standalone simulator
- start Firmware Simulator 2.9
- goto model settings and select external RF module
- select FLYSKY

2. from Companion
- start Companion 2.9
- select File/New then click on +Model
- press cancel to exit the wizard
- double click on the newly created MODEL01
- select AFHDS3 as external radio module
- press simulate

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### Anything else?

Companion W64 build:  Actions, main branch, Windows Companion 64-bit, chore: Translation catchup (#3619) - https://github.com/EdgeTX/edgetx/suites/13479597982/artifacts/740096546

Note: the crash was introduced with commit 125fd2ba (#3442)

@richardclli: please check

richardclli, 2023-06-19:
> Note: the crash was introduced with commit [125fd2b](https://github.com/EdgeTX/edgetx/commit/125fd2ba28ff649a84c7cd117eb694ec3a873fc3) (#3442)
> 

@raphaelcoeffic  The problematic code is your work.  Not sure where is the best way to prevent SIMU to crash in your serial API.

richardclli, 2023-06-19:
Interesting, cannot repeat this with Linux simulator.  It will not crash.

pfeerick, 2023-06-19:
Able to reproduce on WSL2 with current `main` (excuse the Japanese, didn't bother changing it from a prior test build). 

https://github.com/EdgeTX/edgetx/assets/5500713/d8940a6d-e0e0-4d55-b18c-0a366da47650

richardclli, 2023-06-19:
Interesting, if this is the problem, I should be able to repeat in Linux sim, obviously a nullptr problem.

richardclli, 2023-06-19:
@pfeerick Please check if this PR has problem:

https://github.com/EdgeTX/edgetx/pull/3501

@raphaelcoeffic recommended this change and may solve the problem.

https://github.com/EdgeTX/edgetx/blob/97f514fc4dcd68e9207ed926a28f5ec97b684dba/radio/src/hal/module_port.h#L147-L155

pfeerick, 2023-06-19:
Yup, that `nullptr` check will do the trick. So was already fixed in #3501 .

closed 2023-06-19 by richardclli