EdgeTX/edgetx #3686: Can't edit custom Screens on Companion 2.8.4

Fortesque0, 2023-06-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

On new model, you can't edit Custom Screens. No add wigggets, layout, you can't do nothing.

For models imported from radio, you can view custom screens configuration, but you can't modify.

### Expected Behavior

Edit custom screens as you can do on transmitter

### Steps To Reproduce

Create a new model an try to edit custom screens

### Version

2.8.4

### Transmitter

FrSky X12

### Operating System (OS)

Windows

### OS Version

Windows 10

### Anything else?

If it's not a issue, please, improve this functionality to companion.
Thanks

pfeerick, 2023-06-17:
This is a feature that has not been implemented yet - it is view only.

elecpower, 2023-07-24:
Also related to #1270