EdgeTX/edgetx #3688: crossfire 6.19 will not work at 400k

ror1948, 2023-06-17:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

TX16x with edgetx-firmware-nightly-ba56166f
TBS micro Crossfire module V2 with 6.19 and TBS crossfire nano Rx pro
unable to bind with a baudrate of 400k.
It binds at 115K but the output map never populates in the Lua script.
Using LUA script TBS Agent Lite 0.96

### Expected Behavior

bind with baudrate at 400k

### Steps To Reproduce

Tx(micro Tx module V2 installed in bay) 
Press MDL
Press External RF
Select baudrate 400k
Press RTN several times
press SYS button
select TBS AGENT LITE Lua script
Select Tx module
Select bind

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

edgetx-firmware-nightly-ba56166f

ror1948, 2023-06-17:
works as expected with edgetx 2.8.4

pfeerick, 2023-06-19:
I should be able to test this myself also in about two weeks as will be getting a crossfire set then.

3djc, 2023-06-19:
Cannot reproduce (compiled main as of today):

[image]

[image]

[image]

pfeerick, 2023-06-19:
Not so fast... there's two parts to this JC 😆 

> It binds at 115K but the output map never populates in the Lua script.

He wasn't able to bind at 400k, but was at 115k, but then can't do channel/output remapping in the Lua (not sure at what baud rate - it will probably never work at 115k). 

> works as expected with edgetx 2.8.4

Was that @ 400k baud rate?

3djc, 2023-06-19:
like so ?

[image]

pfeerick, 2023-06-19:
That's the one :)

ror1948, 2023-06-19:
Yes all worked at 400k with 2.8.4At 115k output map never populated .But it bound ok.

3djc, 2023-06-19:
All done at 400k, binding and all those screenshots

ror1948, 2023-06-19:
my CF Tx info
[image]

JimB40, 2023-06-19:
Reflash CRSF microTX with same 6.19 and chceck if problem still exist

ror1948, 2023-06-20:
will do
does the version of boot loader matter?
i am using boot loader 2.8.0

ror1948, 2023-06-20:
OK good news

reflashed Tx with 6.19
Flashed RM with bootloader 2.9.0
Flashed RM with nightly Jun 18 2023

set at 400k baudrate

IT ALL WORKS fine
CRSF Rx output map etc populates quickly

Thanks to all

OK to close this issue

pfeerick, 2023-06-20:
Damn... that really is quite annoying... it's intermittent bugs like this that are the most frustrating. Fingers crossed it behaves now! :)

closed 2023-06-20 by pfeerick

JimB40, 2023-06-20:
#@pfeerick just to recap as this issue pops up frequently wit CRSF

1. "115k and output not loading".
Output channeles page have a lot of data 'behind'. One output option needs 3-4 CRSF chunk frames. Other features needs just 1 often. So with 115k which is too slow timeout is thrown because all chunks can't be read. Typical behavior of TBS Agent or Crossfire LUA with too low speed.

2. "Can't connect with 400k"
That is more puzzling. 
This re-flash (refresh) tbs firmware is old trick I've learned writing Agent. Not sure why it behaves like that. But it seems flashing process is not 101% proof.
Nevetheles with serial there are two parties that talk and few weeks ago we had some bug in etx fw that caused problems with CRSF. Seems it is gone now with latest nighlty.

ror1948, 2023-06-20:
it may have been the bootloader version that i was using?
i was using 2.8.0 when i had the problem
now 2.9.0 and it is working.

maybe?

JimB40, 2023-06-20:
You did both at the same time so we can't pin point. You can flash back 2.8.4 bootloader (as this is 10 sec task) and check.
But I'd bet on CRSF FW reflash.

kamik2, 2023-06-21:
https://github.com/EdgeTX/edgetx/issues/3688#issuecomment-1598064542

there are a few posts on facebook etc with the same issue. for some, it loses binding after power cycling the radio on the next day, and some loose binding mid-flight/ ion the ground after some time.
when those people flash their radio back to an older etx version or opentx, the issue is instantly fixed.

to me, it sounds like the model id gets lost/changed with some unknown trigger

ror1948, 2023-06-21:
I did an update to nightly on a friend’s TX16S and it worked at 400k.
I updated bootloader  to 2.9.0 and nightly jun18 2023

- Ralph

pfeerick, 2023-06-22:
> to me, it sounds like the model id gets lost/changed with some unknown trigger

The model id should not change unless you 1) change models (and the new model has a different model id), or 2) change the id manually (or possibly via a Lua script? - but in that case, it's self-inflicted). There is an edge case when switching models with some RF modules, whereby the the model ID is sent before the transmitter module is ready, but this would only result in no connection to the receiver (unless it's id is 0). If you power cycled the radio, or set the model id again once changing models (i.e. change it to something else, and then change it back), it should reconnect instantly. 

Outside of these specific cases, I would instead be suggesting there is a bug in the TBS firmware somewhere.

Ddraig62SPD, 2023-06-24:
Thx for the insight & recommendations guys. Have just migrated from OpenTx 2.3.15 and was getting the exact same issue on all my X9D models. Running V2.8.4 Edge, V6.19 V1/V2 TX, V6.19 Nano RX. 
Reflashed the V6.19 CRSF V2 TX f/w then set 400K baud. Binds were all good and both Output & Channel maps are now displayed in the V0.96 TBS Agent Lite LUA. Much obliged :)

kamik2, 2023-06-25:
> Outside of these specific cases, I would instead be suggesting there is a bug in the TBS firmware somewhere.

that was just an idea why some users could have experienced this issue. - even it would be super weird.
they all had in common, that they switched to etx/ the latest public version of etx, with no update on the crossfire/tracer fw. 

so far I could not reproduce these issues with my old x9d+ on any crossfire fw which makes the bug hunting extremely hard.

ror1948, 2023-06-25:
Take note the last guy switched to 2.8.4 and had the problem.He reprogrammed  TBS TX and all is well.