EdgeTX/edgetx #3691: CLI not working on AUX1 and AUX2

mha1, 2023-06-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

CLI now working on AUX1 and AUX2

Note:
- CLI is working on USB-VCP
- Telemetry Mirror is working on AUX1, AUX2

### Expected Behavior

CLI working on AUX1 and AUX2

### Steps To Reproduce

- Connect serial monitor on either AUX1 or AUX2 GND, RX and TX (115200/8/N, make sure to send cr/lf on enter)
- Select CLI in radio settings hw page on either AUX1 or Aux2
- send help or anything followed by enter (cr/lf)
- observe no CLI output

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

nightly tx16s-c805940.bin

mha1, 2023-06-18:
CLI on AUX1 and AUX2 stopped working with commit 72913eda (PR #3055)

mha1, 2023-06-26:
@raphaelcoeffic please have a look at this. I think the serial driver is missing a vital part for this to work. 

As already stated this stopped working with #3055. 

Changing line 235 of stm32_serial_driver.cpp from `if (!st || st->sp) return nullptr;` to `if (!st) return nullptr;` serial TX using `cliSerialPutc()` works. Why checking st->sp? Checking for already occupied UART? But then why is it occupied with e.g. AUX1/CLI being the only serial consumer.

But still no serial RX. The reason is line 124 in cli.cpp tries to register a callback using the driver function setReceiveCb `if (drv->setReceiveCb) drv->setReceiveCb(ctx, cliReceiveData);` but `setReceiveCb` is initialize in line 494 of stm32_serial_driver.cpp to a `nullptr `and is never set to anything else `.setReceiveCb = nullptr, // TODO`. Check out the comment. I think there is a vital piece of code missing.

I'd appreciate your guidance.

raphaelcoeffic, 2023-07-02:
I probably never tested this :-) Let me have a look to see what's wrong.

raphaelcoeffic, 2023-07-03:
@mha1 the check in `stm32_serial_driver.cpp` is indeed aimed at preventing a serial port which is already used to be re-assigned without prior de-init.

Then, yes, as you found out, `setReceiveCb` is not implemented in the USART driver, only in USB CDC driver (see `usbd_cdc.cpp`). This probably needs to be done so we have something working properly.

mha1, 2024-02-26:
@pfeerick @raphaelcoeffic Doesn't look like this will ever be solved. No big deal, CLI is working via USB-VCP. I'd like to close this with a request to eliminate CLI from the options list on AUX1/AUX2.

pfeerick, 2024-02-26:
Probably best to wait until Raphael resurfaces from work cave before we give up on this idea as it seems like he had thoughts on how to make this happen. But could certainly just kick CLI from the AUX1/2 lists for now though as that mode simply don't work.

mha1, 2024-10-13:
closed with #4679

closed 2024-10-13 by mha1