EdgeTX/edgetx #3692: LUA: LUA system restarted after selecting value in SF Play Track

mha1, 2023-06-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The LUA system is restarted with LUA scripts starting from scratch after selecting a sound file in SF Play Track

### Expected Behavior

LUA system just continuing. I can't see any link from selecting a sound file to LUA that would make restarting the LUA system neccessary.

### Steps To Reproduce

Best to use a 2.9 stand alone simulator (from a recent Companion 2.9) with a PC SD card copy
- copy mini.lua.txt as mini.lua to folder SCRIPTS/FUNCTIONS
- start simulator and enable debug output under View
- edit SF to activate mini.lua as in image 1
- observe debug output for mini.lua message "LUA script mini.lua initialized"
- edit SF to play track and select any sound file
- observe debug output for mini.lua message "LUA script mini.lua initialized"

[mini.lua.txt](https://github.com/EdgeTX/edgetx/files/11780978/mini.lua.txt)
```
local function init()		
  print("LUA script mini.lua initialized")
end

local function run()
end

return { init=init, run=run }
```

Image 1
[image]

Image 2
[image]

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### Anything else?

Here's a shortened down and annotated debug output showing LUA system being closed and started again

```
//
// selected lua script mini.lua to be ON (image #1)
// mini.lua reports being initialized at the end of this section
//
2000.05s: luaInit
2000.05s: luaClose 000001af1bab44c0
2000.05s: lsScripts 000001af1bab6d00
2000.05s: 	not found
2000.05s: f_stat(C:/Users/10137/Desktop/test/SCRIPTS/FUNCTIONS/mini.luac) = error 2 (No such file or directory)
[... 130 lines cut ...]
2059.45s: f_stat(C:/Users/10137/Desktop/test/SOUNDS/en/whlbro.wav) = OK
2059.45s: f_stat(C:/Users/10137/Desktop/test/SOUNDS/en/wngswp.wav) = OK
2059.45s: SCROLL[x=0;y=-330;top=330;bottom=3894]
2059.63s: SD card write labels
2059.63s: f_open(C:/Users/10137/Desktop/test/MODELS/labels.yml, a) = 00007ff91af3fa90 (FIL 00007ff8e13515c0)
2059.63s: f_close(00007ff91af3fa90) (FIL:00007ff8e13515c0)
2063.86s: INDEV_STATE_PRESSED
2063.86s: PRESSED[45|182]
2064.06s: INDEV_STATE_RELEASED

//
// after scanning SOUNDS folder LUA system is closed and restarted
// mini.lua is restarted as a consequence and reports being initialized
// 
2064.15s: luaInit
2064.15s: luaClose 000001af1bab6d00
2064.16s: lsScripts 000001af1bab5d40
2064.16s: f_stat(C:/Users/10137/Desktop/test/SCRIPTS/FUNCTIONS/mini.luac) = OK
2064.16s: f_stat(C:/Users/10137/Desktop/test/SCRIPTS/FUNCTIONS/mini.lua) = OK
2064.16s: luaLoadScriptFileToState(/SCRIPTS/FUNCTIONS/mini.lua, T): loading /SCRIPTS/FUNCTIONS/mini.lua
2064.16s: f_open(C:/Users/10137/Desktop/test/SCRIPTS/FUNCTIONS/mini.lua, 1) = 00007ff91af3fa90 (FIL 0000004db27fea08)
2064.16s: f_close(00007ff91af3fa90) (FIL:0000004db27fea08)
LUA script mini.lua initialized

```

philmoz, 2023-06-18:
In file gui/colorlcd/special_functions.cpp, line 195 calls LUA_LOAD_MODEL_SCRIPTS() which restarts Lua and reloads all scripts. 

This happens for PLAY_SCRIPT, PLAY_TRACK and BACKGND_MUSIC special functions.

It should probably only be called if 'func == FUNC_PLAY_SCRIPT'.

mha1, 2023-06-18:
@philmoz Thanks for your quick response. I think your suggestion makes a lot of sense. PLAY_SCRIPT is the only one interacting with LUA at all.

BTW: nothing to do with this. I have a LUA script that uses model.setCustomFunction() to enable and disable a SF. Basically I use this to get the volume to MAX if required by the LUA script and back to the volume knob settings. Watching the SF page while the script is running the line for the SF that I enable/disable reflects this in 2.8 so you can watch it changing state (line highlighted/not highlighted). 2.9 doesn't do that and I presume it is because it doesn't act on external changes of the enable status.

[image]

philmoz, 2023-06-18:
> BTW: nothing to do with this. I have a LUA script that uses model.setCustomFunction() to enable and disable a SF. Basically I use this to get the volume to MAX if required by the LUA script and back [...]

I can't reproduce this with current main branch. The SF lines update as expected in the SF list view if the enable state is changed by a Lua script.

closed 2023-06-19 by pfeerick

mha1, 2023-06-19:
Thanks for fixing this. And thank you very much for trying the LUA enable. I repeated it with my script and it is working. Strange, I blame squirrels.

PR #3698 (2023-06-19): fix(color) - Only reset Lua if changing PLAY SCRIPT special function
Don't reset Lua for PLAY TRACK, or BACKGND MUSIC functions.

Fixes #3692