EdgeTX/edgetx #3695: 2.9 - When writing the ELRS firmware on TX the sentence is wrong

robustini, 2023-06-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I just noticed today that when I write the firmware of the FrSky R9M module with ELRS during the flash at the bottom the words "module reset" remains even while writing as seen in the image, when it should only appear initially and then write "writing in progress".

### Expected Behavior

It should do what I wrote above, I tried it with an older EdgeTX firmware (2.8) and in fact it works.
I don't know if the problem is in the language, I haven't tried in ENG, only IT.

### Steps To Reproduce

1. Flash the latest master (2.9)
2. Write ELRS firmware to external module

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

### Operating System (OS)

Windows

### OS Version

Windows 10 Pro x64

### Anything else?

[image]

pfeerick, 2023-06-19:
It's not unique to ELRS... it's the same for all. We'd noticed it before, but I didn't realise it was correct in 2.8, so will see what changed there - thanks for pointing that out! :)

robustini, 2023-06-19:
@pfeerick thanks Peter for your feedback.
In the TX16S I have firmware compiled in November 2022, actually quite dated, and it works there.

PR #3704 (2023-06-22): fix: flash progress text stuck at device reset
fixes #3695
commit d5796191 (2023-06-22): fix(color): Firmware update progress text stuck at device reset (#3704)

* fixes #3695 flash progress

* move init() to where it belongs (constructor)
commit 470db437 (2023-06-23): fix(color): Firmware update progress text stuck at device reset (#3704)

* fixes #3695 flash progress

* move init() to where it belongs (constructor)
closed 2023-06-22 by pfeerick