EdgeTX/edgetx #3696: SF with no switch assigned becomes ghost function for life cycle and isn't written to .yml

mha1, 2023-06-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Creating a SF and leaving the Switch at the "---" default will not be correctly displayed after creation and will not be stored permanently.

### Expected Behavior

SF should be visible in SF list and permanently stored

### Steps To Reproduce

- go to next available free SF an click to create new SF
- edit like in attached image
- leave edit window
- observe SF not populated
- click on the same SF to edit (even if empty)
- observe SF still present in the system but not listed in used SF list

[image]

### Version

2.8.3

### Transmitter

Radiomaster TX16S / TX16SMK2

pfeerick, 2023-06-19:
This is a bit of a weird one... 

At present if you don't assign a trigger/switch... the special function effectively does not exist. Perhaps the UI could demonstrate this better by hiding all the other elements until the switch condition is changed away from `---` (and also set back to that if you try to be "smart" and set it away from and then back again)? It would be a bit different when thie

While 2.9 has a completely different adding mechanism... it still has this "remember how you last configured the SF dialog" hangover... not sure whether to consider this as an actual bug or a "save your from yourself stupidity when you cancel by mistake" feature 😆 With #3024 in place, this would IMO definitely be a bug - i.e. it should be saved, but you can't enable it if there is no trigger defined.

mha1, 2023-06-20:
Understood, but If you consider the current behavior a feature consequently the unfinished SF should show up in the SF list and be written to yml until there is a different solution.

pfeerick, 2024-02-26:
Only if this is still reproducible post #3024 - that PR should have prevented this being possible IIRC

mha1, 2024-02-27:
@pfeerick still reproducible in 2.9 and 2.10

mha1, 2024-07-06:
@pfeerick hooray, we made the 1yr mark