EdgeTX/edgetx #3697: Make Play Sound playable in Companion

rufus-se, 2023-06-18:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I'm new to EdgeTX and I love to work in companion. I am missing that I can´t hear the built in sounds. Frustrating to have to use the radio just to listen to a sound. I would like Play Sound to work the same way as Play Track. 

### Describe the solution you'd like

Sounds should be playable in Companion in the same way as Tracks. 
[image]

rufus-se, 2023-06-18:
New to GitHub as well. Tried to add a companion label but don´t know how to...

elecpower, 2023-06-18:
@rufus-se the sounds are not read from the sd card but real time generated on the radio. Due to the effort required and the fact that they can be played in the simulator, this feature is unlikely to be implemented in Companion.

pfeerick, 2023-06-19:
I agree as far as effort vs gain... especially as using the simulator is the current option. 

Just a thought though... could we package the sounds up as a bunch of wavs, and play them that way? Would that make it easier, or was that already what you had in mind as the effort required?

elecpower, 2023-06-19:
@pfeerick creating wav files and bundling in the standard sd card images SYSTEM folder did cross my mind. This would require changes to the sd card image builder script as the sounds are not language specific. Then there are the custom sound packs for which we have no control.

An alternative and likely a better option would be to generate the wav files (once only task) and bundle into the Companion resources as part of the exe compilation process. The SF play button would access the resources. With this option no need to change the sd card build processes and no risk they are not found.

rufus-se, 2023-06-19:
After got this explained I totally understand the effort vs gain reasoning. It also explains why I haven't been able to find those sound files. 
I didn't think about having them as wavs. That could of course be an option. No idea about pros and cons between wavs and radio generated sounds. I guess a consequence would be that all "sounds" become Tracks. This could actually be good since a lot of people confuses Tracks and Sounds. And when you download a new soundpack you get a bunch of tracks.

pfeerick, 2023-06-19:
> An alternative and likely a better option would be to generate the wav files (once only task) and bundle into the Companion resources as part of the exe compilation process. The SF play button would [...]

This is what I had in mind... there's no befit/gain to it being part of the sd card pack IMO... it's a Companion side resource. I was wondering if you thought that was a painless enough solution to the problem ;)

"Play Sounds" had a place more in OpenTX when not having a SD card was a possibility... not so much now since it is a requirement. However, it does conveniently separate beep and tone type audio from mostly(all?) spoken audio. So there could be an argument now for rolling to two options together, but I think that is a conversation for later?

3djc, 2023-06-19:
Well, it avoids having a bunch of more wav file in the large list already, you could have them in a distinct folder I guess, but then thats more radio sd io for only companion benefit ?

elecpower, 2023-07-24:
If someone has the utilities to generate the sound files consistent with voice files eg volume, etc then I am agreeable to plugging them into Companion.

jrwieland, 2023-07-25:
@elecpower  OpenTX used to include the csv file within the sounds system folder that contained all of the system sounds which could then be used in TTS programs.  I have maintained that list for use with edge and kept it updated as it is has progressed.
You can generate several TTS voices using TTSAutomate (link below).

As far as volume control I like my system sounds louder than any user created tracks (music, model name, etc.) that way any warnings (RSSI, altitude, battery voltage, etc.)  announce clearly over the user tracks.  I set the volume to 1.9 using  Pazera Free Audio Extractor (link below) it can convert many formats to wav as well as several others,  but it does a better job at setting batch volume wav to wav conversion than Audacity does.

Here is the cvs file with all of the current system sounds 2.8
[Edgetx.csv](https://github.com/EdgeTX/edgetx/files/12165057/Edgetx.csv)

links to the files mentioned above
https://ttsautomate.com/
https://www.pazera-software.com/products/audio-extractor/

pfeerick, 2023-07-26:
We're not talking about the TTS sounds @jrwieland, but the Play Sound system generated sounds - like the chirps and beeps. For Play Track stuff, Companion already uses the SD Card Pack folder as defined in the radio profile settings, to use the same sound files the radio uses. But hopefully that info is of help to someone else :)

For this, unless someone has a better idea, *someone* would need to walk through each of the sounds one by one on a transmitter, and record the sounds (i.e. maybe using a transmitter that has a headphones jack/decent audio path and record that via line-in).

jrwieland, 2023-07-26:
Ok...I was confused apparently.  That also explains why it wasn't an issue in the sounds repository.

closed 2023-09-04 by pfeerick