EdgeTX/edgetx #3702: Boot animations

ggoraa, 2023-06-20:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Pretty simple for both platforms:
- B&W: a custom video format that compresses each frame into changes on top of the previous frame. I implemented this in my EdgeTX Bad Apple project, so I have experience with implementing those kinds of things
- Color: Or a normal MPEG (or smth else) video if LVGL is initialized before showing the boot image, if not, then something similar to the B&W approach

### Describe the solution you'd like

d

philmoz, 2023-06-20:
B&W radios already have a startup animation.

I don't see any value in adding this to color radios as it would only serve to increase the startup time even more.

pfeerick, 2023-06-20:
While that is true... some people want exactly that ;) https://github.com/EdgeTX/edgetx/issues/2537

pfeerick, 2023-06-20:
Also keep in mind on B&W, the splash screen is shown before the SD card is mounted (hence why it is embedded in the firmware).

ggoraa, 2023-06-20:
In "boot animation" I mean a video, not a static image, so it's not a thing right now

ggoraa, 2023-06-21:
> So the SD card is available, it's just that (S)RAM is a problem. 

Actually, that's why I suggested that custom format, because this is the exact problem I've run into when making the Bad Apple video play, there is simply not enough RAM to store even two frames in Lua, it should be optimiser as much as possible. When doing that with C/C++, it can be optimised much further, making it a viable option to implement

JimB40, 2023-06-21:
... and what about simple thing like nice bootloader background? :)
This was closed (don't know why) https://github.com/EdgeTX/edgetx/issues/638

[image]
[image]

I won't beat around the bush. 
Current background looks like Old-Rotten-Swiss-Cheese, not mention distracting from data.
[image]

pfeerick, 2023-06-21:
> I won't beat around the bush.

I won't either... which PR is linked to that issue / closed it? It is there at the bottom of the issue... you *DO* know why it was closed... 😝 I think the problem probably was that while it included the resources from the issue, the PR focused on the splash screen more than the booloader, so don't completely resolve that PR 🙃

ggoraa, 2023-06-23:
Does EdgeTX have LVGL loaded before showing the boot image?

ggoraa, 2023-06-25:
@pfeerick I have made a document outlining the file format. Do you have any suggestions/edits that you would like to share?
https://www.craft.me/s/BRzGbfKdcQirmd