EdgeTX/edgetx #3706: You can left "empty" source for an imput but you can´t select "---"(empty) on simulation or radio

Fortesque0, 2023-06-21:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Simulator

### Current Behavior

For an input, you cant left "empty" source data. When you edit this imput on simulation or on radio, you have not "empty" choice. 

### Expected Behavior

Has "empty" selection on simulation and radio to.

### Steps To Reproduce

On companion, edit or create new imput
Let empty source
Start simulation or transfer to radio
Go to simulation/radio imput menu and edit input.
Select one kind of source or try to find "---" if you select some source and accept, you can´t let empty again.

### Version

2.8.4

### Transmitter

FrSky X12

### Operating System (OS)

Windows

### OS Version

Windows 10

### Anything else?

[image]
[image]

Fortesque0, 2023-06-21:
[image]

philmoz, 2023-06-21:
I think the issue is that Companion allows the source to be empty when it should not do so.

elecpower, 2023-06-21:
Empty source can also occur in radio conversions where the source does not exist on the current profile radio.
Pilots want the row kept so all they have to do is fix the source and not have to re-enter all the other details.

elecpower, 2023-06-21:
Also empty source required for copy and paste and drag and drop between multiple open model and settings files.

Fortesque0, 2023-06-22:
Hello,
I am new user and my be I am wrong, but, I think empty source it's absolutely necessary. 
I need a constant value (GVAR) active with switch position, and assigned to an output.
@

Fortesque0, 2023-06-22:
Sorry, y closed by mistake

pfeerick, 2023-06-23:
> I need a constant value (GVAR) active with switch position, and assigned to an output.

Why though? An input *needs* to have a source - this is why the radio won't let you set it, but Companion just happens to let you (for its own technical reasons).  Why not set it to MAX, and then work from that known state? 

Maybe explain what it is you're trying to achieve, and we can see why you think empty needs to be an option. Perhaps you're approaching the problem you're trying to solve the wrong way? Or you're seeing a use case we haven't thought of. i.e. if you're trying to make it so, a 3POS switch has one value in the middle position, and another when it is up and down, I do something similar, but I use a curve instead to control that. No need for the GVs, no need for an "empty" input source. The switch used as the input source, and I use the curve to change what is done with that input.

Fortesque0, 2023-06-24:
Hello,
The functionality it's set giro values for helicopters.
I want to assign to output 5 two different values using one interrupt. This values can be increased or decreased (linked to interrupt position) using trim.
For me, source selection must be included GV as option, because source philosophie should be any input.
You can see on my last uploaded  image this configuration for Input 5 Gyro.
Regards

3djc, 2023-06-24:
MAX with weight in GV is the way to do it

Fortesque0, 2023-06-24:
😱😱 I say I am new😅😅 thanks in advance, I will try it.

Fortesque0, 2023-06-25:
Hi,
Yes, using MAX and modifying weight  with GVAR I can do it.
But, as philosophy, I think GVAR should be a source input too as MAX is. 
MAX it's a 100% value. GVAR can be any value, included 100%.
Please, considere this option.
Thank you for your help and developers, close this issue if you considered it's done.
Nice flights!!

pfeerick, 2023-07-04:
To do that, you would use GV as the source for an INPUT, and then use that input in a mix. You have a wider selection of sources when you define stuff at the input stage. ;)

closed 2023-07-04 by pfeerick

Fortesque0, 2023-07-13:
> To do that, you would use GV as the source for an INPUT, and then use that input in a mix. You have a wider selection of sources when you define stuff at the input stage. ;)

Hello pfeerick,

Perhaps i am wrong ,but, you say that i can select GVAR as the source for an INPUT??? Just i am saying that there is not choice to select a GVAR  on input menu.. i think it´s good idea to select GVAR as an input source, like MAX is,

[image]

mha1, 2023-07-13:
Deselect all icons on the left side for the full list and scroll down.

[image]

Fortesque0, 2023-07-14:
Hello MHA1,

Thaks for your reply. I have been testing again and i have found some discrepances. 

1.- Yes, you can deselect icons and GVx appears, but without "labeled name"
[image]

1.1-.....but, using companion to program your radio, you haven´t this option:
 
[image]

2.- GVAR are labeled if you select ass weight or offset
[image]

3.- You don´t think should be better to include labeled GVx with MAX? 

[image]

So i think this should be reopen. My be on a new one issue?
What dou you think Team?
Regards

mha1, 2023-07-15:
2.9 Companion will let you define GVARs as Input:
[image]

As for the label I recall a discussion about that. Maybe @philmoz can speak about the GVAR label status and your proposal to cluster GVARs with MAX.

pfeerick, 2023-07-16:
The question of whether to group GVAR with MAX really depends on *what* that category was meant to organise... i.e. GVARS probably should be in a category of their own. I did a PR to add the GVAR names if set, but put that off until 2.10. 

> So i think this should be reopen. My be on a new one issue?

If you're talking about something different to the initial issue (which you are, aren't you?), it belongs in a new issue ;) But I think from M's replay you can already see you can do it in Companion, just hadn't scrolled down far enough (looks like they're not in exactly the same order).

mha1, 2023-07-16:
2.9rc1 firmware and Companion allow to have GVAR's as input. Nothing to do here I guess. If GVAR labels are a low hanging fruit for 2.9 why not. No issue required, am I right?

pfeerick, 2023-07-16:
no, No, nO, NO! No more substantive changes to 2.9! I said this over a
month ago! 🤣🤣

mha1, 2023-07-17:
Time's flying, I remember it just like yesterday. I suppose it's not that low hanging then.