EdgeTX/edgetx #3711: Incorrect display of six position switch operation

Shouldercannon, 2023-06-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The six-position switch shows the wrong values in the positive direction.
[image]
[image]

### Expected Behavior

It should be like this
[image]
[image]

### Steps To Reproduce

1. Push Sys
2. Go to the sixth tab of the menu
3. Go to Anas
4. Switch to position 4, 5, 6.

### Version

2.8.4

### Transmitter

Radiomaster Boxer

### Operating System (OS)

Windows

### OS Version

Win10 x64 22H2

### Anything else?

This problem has been observed for a very long time.

philmoz, 2023-06-22:
What are the actual RAW numeric values for A7 at each position?

pfeerick, 2023-06-23:
I see this on mine also -> 

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

philmoz, 2023-06-23:
Does the 6POS switch have to be calibrated on this radio?

pfeerick, 2023-06-23:
No, but even after calibration it is out slightly.

pfeerick, 2023-06-23:
I doubled the last screenshot I noticed A6 was still showing 100 regardless of whether it was raw value of 2046 or 2047... 🤔

philmoz, 2023-06-23:
What are the RAW values?

pfeerick, 2023-06-23:
[image]
[image]
[image]
[image]
[image]
[image]

philmoz, 2023-06-23:
Thanks. My TX16S does the same :(

pfeerick, 2023-06-23:
It's also present in main, so unless the ADC refactor branch happens to touch on this somehow, it's still a current bug. 

@3djc Any ideas? Maybe `src/switches.cpp` -> `getSwitchesPosition()` is off by one past the zero or something when determining the steps? As it seems a bit od it is off by one for the last three positions

philmoz, 2023-06-23:
It looks like a rounding problem with the way the 6POS values are being calculated - still trying to understand all the code though so could be wrong.

3djc, 2023-06-23:
I'll have a look in ADC branch, we reviewed some of that (but not specificaly looking at that)

pfeerick, 2023-06-23:
I just had a quick look, and the equivalent lines changed in #3713 appear to be the be the same there

https://github.com/EdgeTX/edgetx/blob/c8e9437cd23349630de3c6fb4e5e4fd2c57fe177/radio/src/hal/adc_driver.cpp#L403-L413

So unless it is fixed elsewhere, if you're happy with the fix in #3713, I can merge that into 2.9 only, and you can apply it in the ADC branch once Malte pushes that rebase...

3djc, 2023-06-23:
Sure sure

philmoz, 2024-08-04:
Can this be closed?

pfeerick, 2024-08-04:
I think so ... my Boxer is showing all correct values on 2.10.2, so well past the prospective fixes being merged.

PR #3713 (2023-06-25): fix(radio) - Fix rounding errors in ADC value calculation for 6POS switch.
Fixes #3711 

Tested in simulator and on TX16S Mk2.
closed 2024-08-04 by pfeerick