EdgeTX/edgetx #3722: Digits (only) keyboard for numeric inputs

knoppixmeister, 2023-06-26:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

It would be nice to have digits (only) keyboard for numeric input fields like it made e.g. in Android.

example:

[image]

### Describe the solution you'd like

Create digits (only) keyboard.

ggoraa, 2023-07-05:
You know what, this is a great suggestion! I can’t think of a reason to _not_ implement this keyboard layout variant

J-Sorenson, 2025-09-25:
...and a decimal point.

philmoz, 2025-09-26:
The complexity is that the numeric input fields all have range limits (min & max values).
This is difficult to manage / enforce if users can enter arbitrary values.