EdgeTX/edgetx #3725: LiteRadio 3 Pro - not compiling Betaflight script

odixxx, 2023-06-28:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

when trying to compiling Betaflight Setup script - Syntax Error
[image]

### Expected Behavior

My setup:
Betaflight TX Lua Scripts 1.7.0
Betaflight 4.4.2
EdgeTX v2.8.4

ps. on another radio X9D Plus everything works fine!

### Steps To Reproduce

Syntax Error

### Version

2.8.4

### Transmitter

BetaFPV LiteRadio3 Pro

### Operating System (OS)

Windows

### OS Version

11

pfeerick, 2023-07-01:
I've just tried this on v2.8.4, with a fresh copy of the Betaflight Lua v1.7.0, and it compiled just fine. Given the actual error is "file not found", have you tried copying the betaflight lua pack onto the radio again (and making sure you "safely remove" the radio USB device before unplugging the USB) just in case not all the files were copied on properly?

odixxx, 2023-07-01:
I just tried formatting the card and everything is new - without result!
can you give me your file - bf.luac

pfeerick, 2023-07-03:
[BF-SCRIPTS.zip](https://github.com/EdgeTX/edgetx/files/11935946/BF-SCRIPTS.zip)

This is my SCRIPTS directory from the LR3P minus extra baggage.

odixxx, 2023-07-03:
thanks, everything worked!!!
in the files "betaflight-tx-lua-scripts_1.7.0.zip" - apparently something is missing

pfeerick, 2023-07-03:
Something strange going on anyway... that is what I used when trying to repo this. Anyway, glad it's working for you now. 

[image]

closed 2023-07-03 by pfeerick

VisualGadget, 2023-10-08:
I had the same error on a different TX and problem was the faulty microSD card.