EdgeTX/edgetx #3732: v1 in logical switches drops to random parameters

ivan-ef, 2023-06-28:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

v1 in logical switches drops to random parameters after transmitter reboot. 
it may change to T6+/- or T5 or L1 etc.

[image]

[image]

### Expected Behavior

Correct v1 parameters in logical switches after transmitter reboot.

### Steps To Reproduce

1.set V1 of logical switch to rudder trim + and -
2. reboot.

### Version

Other (Please specify below)

### Transmitter

Jumper T-Pro

### Operating System (OS)

macOS

### OS Version

monterey

### Anything else?

[model02.yml.zip](https://github.com/EdgeTX/edgetx/files/11900515/model02.yml.zip)

Model file may be corrupted during connection of transmitter in sd-card USB storage mode. Transmitter drops USB connection often and just turns off. Can't eject USB storage correctly on Mac, it says storage is used in few apps. 

Stock firmware version:
[image]

3djc, 2023-07-03:
Please try lastest nightly, I think I fixed this some time ago

3djc, 2023-07-03:
Tested your file, not showing the issue, please update your firmware version with a nightly

closed 2023-07-03 by 3djc

ivan-ef, 2023-07-03:
Updated to nightly, looks ok now.