EdgeTX/edgetx #3736: 2.9: SD card wrie error "disk_write bad alignment"

mha1, 2023-06-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Building 2.9 (commit a7f1b674) in debug configuration reveals a potential problem with writing to SD card that probably needs to be understood.  At least I don't know if this is expected or not.

radio\src\targets\horus\diskio.cpp:
[image]

Looking at the trace below the pointer `const Byte *buff` to the data to written seems to point to an address not in RAM (starting 0x2000000) triggering some recovery action. The write seems to finish ok.

Maybe someone with better insight into FatFS can check.

16.05s: TE_NONE
16.80s: eeprom write general
16.80s: YAML radio settings writer
16.84s: disk_write bad alignment (0x100083b8)
16.84s: disk_write bad alignment (0x100083b8)
16.85s: disk_write bad alignment (0x100083b8)
16.85s: disk_write bad alignment (0x100083b8)
16.85s: disk_write bad alignment (0x100083b8)
16.86s: disk_write bad alignment (0x100083b8)
16.86s: disk_write bad alignment (0x100083b8)
16.97s: generalSettings written with checksum 65164
16.99s: SD card write labels
17.18s: eeprom write model
17.18s: YAML model writer
17.19s: disk_write bad alignment (0x100082d0)
17.23s: disk_write bad alignment (0x100082d0)
17.23s: disk_write bad alignment (0x100082d0)
17.24s: disk_write bad alignment (0x100082d0)
17.24s: disk_write bad alignment (0x100082d0)
17.24s: disk_write bad alignment (0x100082d0)
17.25s: disk_write bad alignment (0x100082d0)
17.25s: disk_write bad alignment (0x100082d0)
17.25s: disk_write bad alignment (0x100082d0)
17.26s: disk_write bad alignment (0x100082d0)
17.26s: disk_write bad alignment (0x100082d0)
17.26s: disk_write bad alignment (0x100082d0)
17.27s: disk_write bad alignment (0x100082d0)
17.27s: disk_write bad alignment (0x100082d0)
17.27s: disk_write bad alignment (0x100082d0)
17.28s: disk_write bad alignment (0x100082d0)
17.28s: disk_write bad alignment (0x100082d0)
17.28s: disk_write bad alignment (0x100082d0)
17.29s: disk_write bad alignment (0x100082d0)
17.29s: disk_write bad alignment (0x100082d0)
17.29s: disk_write bad alignment (0x100082d0)
17.30s: disk_write bad alignment (0x100082d0)
17.30s: disk_write bad alignment (0x100082d0)
17.30s: disk_write bad alignment (0x100082d0)
17.31s: disk_write bad alignment (0x100082d0)
17.31s: disk_write bad alignment (0x100082d0)
17.31s: disk_write bad alignment (0x100082d0)
17.35s: <TESTHoTT/0> : 6,38,4
17.35s: <TESTHoTT/1> : 0,0,1
18.35s: RamBackupWrite sdsize=16406 backupsize=4656 rlcsize=459
32.35s: SD card write labels<\r>

### Expected Behavior

don't really know but feel like the trace is trying to tell a story

### Steps To Reproduce

build 2.9 in debug configuration
connect serial monitor to AUX1 and set AUX1 to DEBUG

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S / TX16SMK2

mha1, 2023-06-30:
2.9 does this for yaml writer and logs. 2.8 also traces bad alignment but only for yml writer, not for logs. 

Maybe this is a feature rather than a bug. I'd appreciate someone enlightening me.

philmoz, 2023-06-30:
To me it just looks like the code is ensuring that all SD card writes are done from SDRAM and are from a buffer that is DWORD aligned. If this is not the case the data is copied to a safe buffer to do the actual writing.

The 0x1000xxxx addresses are in the stack memory area, so this will be code that is writing data from stack variables instead of SDRAM variables.

Personally I would just get rid of the TRACE statement as it is misleading.

mha1, 2023-07-01:
You are right, I didn't think of CCM. It's not a bad idea to have the stack in CCM. It is as little faster than SDRAM but CCM has one side effect. As it has a separate bus directly wired to the core (core coupled memory) it can't be used for DMA. That's why this piece of code copies stack allocated buffers to SDRAM and also ensuring dword alignment.

The trace message is really misleading. Got me for sure.

mha1, 2023-07-06:
Misleading trace message, firmware ok.

closed 2023-07-06 by mha1