EdgeTX/edgetx #3740: 2.9 Nightly - The first 4 channels seem to go crazy, sweeping up to full scale randomly

robustini, 2023-07-01:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I think a video is enough to see what is happening, obviously I am not giving any commands to the sticks.
Flashed a version compiled on June 22 and no problems.

https://github.com/EdgeTX/edgetx/assets/3864712/11525aeb-e3dc-4661-aee2-65d343ac6868

### Expected Behavior

Of course, this should not happen, the channels should remain stable at the set value.

### Steps To Reproduce

1. Flashes the latest firmware compiled from the [July 1 main branch](https://github.com/EdgeTX/edgetx/commit/632eceba6d2daaa96df4f530b8275ae67e38f4d7).
2. Go to channels monitor

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

### Anything else?

I have not noticed any malfunctions in the other channels from 5 onward.

gagarinlg, 2023-07-01:
The nightlies as 2.10 2.9 is branches off and should get RC1 any day now

Talha909, 2023-07-01:
I have the same issue with Frsky Xlites Pro. All channel are going arond here and their. Calibration is also not wokring. S1 S2 swtiches are gone. I cant able to see these two switces

gagarinlg, 2023-07-01:
Which modes (1,2,3, 4) are you on?

koaxheli, 2023-07-01:
I also have the same problem, switches S1...S2 are gone.
The selection field for sliders is completely missing and when I press on the selection field Potis the transmitter goes to EMERGENCY MODE.

Version is 2.10.0 nightly (8f431eaf)

Transmitter is TX16S

What else is noticed that all 3 timers start immediately,
although the setting is on throttle control.
[screen.zip](https://github.com/EdgeTX/edgetx/files/11926576/screen.zip)

pfeerick, 2023-07-02:
> What else is noticed that all 3 timers start immediately, although the setting is on throttle control. [screen.zip](https://github.com/EdgeTX/edgetx/files/11926576/screen.zip)

Most likely needs this https://github.com/EdgeTX/edgetx/pull/3733

Talha909, 2023-07-03:
I am in mood 2. There is also one trim button disappared from the display.

pfeerick, 2023-07-03:
Since you have the XLite S Pro, can you give https://github.com/EdgeTX/edgetx/pull/3745 a try? That should fix the stick issue.

koaxheli, 2023-07-13:
Still EMERGENCIMODE on FW v2.10.0 as the LS and RS sliders are moved to the pot selection field.
[image]

In FW v2.9.0 the switch fields Poti and Slider were separated.
[image]
[image]

In the nightly version everything is OK, in the selfbuild the error occurs.
[image]
[image]

raphaelcoeffic, 2023-07-14:
@koaxheli sliders is `main` branch are indeed no different anymore from pots. Are you sure this is related to the issue described above? What is you actual issue? The crash-reboot? (Emergency Mode) We'll need quite some more details on what you're actually doing, which radio, etc before we can do anything about it. But again, this seems like a different issue to me.

closed 2023-07-14 by pfeerick

koaxheli, 2023-07-14:
My problem is when I create the new FW v2.10.0 via Github the slider button is missing as it is now under Potis.this is new, but when I then want to open the button Potis the transmitter goes into emergency mode.Transmitter is TX16S Radiomaster which until FW v2.9.0-rc1 everything was always OK. Github compilation is the following. Gesendet von Mail für Windows Von: Raphael CoefficGesendet: Freitag, 14. Juli 2023 06:40An: EdgeTX/edgetxCc: koaxheli; MentionBetreff: Re: [EdgeTX/edgetx] 2.10 Nightly - The first 4 channels seem to go crazy, sweeping up to full scale randomly (Issue ***@***.*** sliders is main branch are indeed no different anymore from pots. Are you sure this is related to the issue described above? What is you actual issue? The crash-reboot? (Emergency Mode) We'll need quite some more details on what you're actually doing, which radio, etc before we can do anything about it. But again, this seems like a different issue to me.

pfeerick, 2023-07-14:
> which until FW v2.9.0-rc1 everything was always OK

Did you mean nightly here? As FW v2.9.0-rc1 does not have any of these changes and doesn't seem to have any issue with the pots or sliders. And if you are using a nightly, and then revert back to 2.9-rc1, settings are likely to be corrupted as newer versions of EdgeTX are generally backwards compatible, not forward compatible. The nightlies/2.10 development versions are expected to be somewhat unstable atm...