EdgeTX/edgetx #3746: Companion File Compare Issue

rdeanchurch, 2023-07-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

1.In EdgeTxCompanion open an existing .etx that has a model setup.  
2. Open a new blank .etx.
3. Copy a model from 1 and paste to 2.
4. . perform a' File Compare' between the model in etx 1 and the model in etx 2
5.  All instances of switches will be SwX in 1 and SX in 2,  ( ISSUE! )
       so they will show as differences in the 'file compare' window.
6. they both execute as expected in the EdgeCompainion Simulator.

### Expected Behavior

The two models compared should be identical.

### Steps To Reproduce

See Current Behavior

### Version

2.8.4

### Transmitter

Radiomaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### OS Version

Win 11on a Dell Inspiron

### Anything else?

[image]
[image]

elecpower, 2023-07-06:
Check the Radio Settings for each etx. You should find the source has label text overrides and the destination does not and thus the reason for the differences

rdeanchurch, 2023-07-08:
My bad, I missed that.  
Please close this issue.

closed 2023-07-08 by rdeanchurch