EdgeTX/edgetx #3751: Log interval Changes in global functions are not saved

ParkerEde, 2023-07-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

With the 2.9.0 RC1 it is currently no longer possible to change the LOG interval time from e.g. 1.0s to 0.5s in the global functions, as this is not saved. At first everything is displayed correctly, but when you leave the system menu and open it again, then open the global functions, the value is back to the initial value after you have gone back into editing.

### Expected Behavior

the value should be saved

### Steps To Reproduce

1. open SYS
2. open GF
3. change intervall value for SD-LOG
4. close GF
5. close SYS
6. open SYS
7. open GF
8. edit entry for SD-LOG
9. observ the value is not saved

### Version

Other (Please specify below)

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

### Anything else?

2.9.0 RC1

pfeerick, 2023-07-03:
Happens for both special functions and global functions for me on TX16S

pfeerick, 2023-07-03:
Most likely caused by #3058 as I think

https://github.com/EdgeTX/edgetx/blob/8f0ad3fca829f68a0051f12708e29dfba969ec86/radio/src/gui/colorlcd/special_functions.cpp#L235

obliterates the current value every time the dialog is opened 🤭

PR #3755 (2023-07-04): fix: SD Logs interval shown as default value when entering SF edit window
fixes #3751

changes:
- use stored value if SF exists otherwise initialize with default value
closed 2023-07-04 by pfeerick