EdgeTX/edgetx #3752: BT Settings missing in Hardwaremenu for Taranis QX7S

ParkerEde, 2023-07-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Compared to the QX7, the Taranis QX7S has a built-in Bluetooth module. However, this is not displayed under SYSmenu/Hardware for configuration. This means that it cannot be activated for the trainer function and cannot be selected in modelsettings under DSC BTMaster or BTSlave.
This setting is already offered in Companion.

With the Taranis QX7S ACCESS, on the other hand, it is already correct.

### Expected Behavior

QX7S should offer BT settings in Hardwaremenu

### Steps To Reproduce

1. Enter Hardwaremenu
2. try to edit BT settings
3. observ, the BT settings are not offered

### Version

Other (Please specify below)

### Transmitter

FrSky Q X7 / Q X7S (ACCST)

### Operating System (OS)

Windows

### Anything else?

2.9.0 RC1

pfeerick, 2023-07-03:
Was this the case in 2.8 also?

ParkerEde, 2023-07-03:
I have never tested this before, as I have only had the radio since yesterday. If the test is of great importance, I can still do it.

pfeerick, 2023-07-03:
Ah, ok. I suspect it was then, as I don't think some elements of Bluetooth have worked properly for a while now. The only "importance" is in trying to determine whether it is new or pre-existing behaviour ;)

ParkerEde, 2023-07-03:
Okay. The BT settings are also missing in 2.8.4. I have just checked it. So it seems to have always been missing.

ParkerEde, 2023-07-03:
That was my first thought too and I tried it out with that. Unfortunately, that doesn't work either.

`cmake -G "MSYS Makefiles" -Wno-dev -DCMAKE_PREFIX_PATH=$HOME/5.12.9/mingw73_64 -DSDL2_LIBRARY_PATH=/mingw64/bin/ -DPCB=X7 -DDEFAULT_MODE=2 -DBLUETOOTH=YES -DGVARS=YES -DHELI=YES -DLUA=YES -DINTERNAL_GPS=NO -DTRANSLATIONS="DE" -DCMAKE_BUILD_TYPE=Release ../`

pfeerick, 2023-07-03:
What is *actually* needed, is for this to finally happen https://github.com/EdgeTX/edgetx/issues/1390. i.e. the runtime configurable bluetooth that has been waiting to happen.

3djc, 2023-07-03:
Code seems sound : https://github.com/EdgeTX/edgetx/blob/8f0ad3fca829f68a0051f12708e29dfba969ec86/radio/src/targets/common/arm/stm32/bluetooth_driver.cpp#L130

philmoz, 2023-07-03:
The function BT_USART_IRQHandler in bluetooth_driver.cpp is not compiled into the firmware (#if _OBSOLETE_). So the btChipPresent flag can never be set.

For X7ACCESS the IS_BLUETOOTH_CHIP_PRESENT macro defaults to true so does not need the btChipPresent flag.

3djc, 2023-07-03:
ok, so another victim of the serial rework :) Should be an easy fix

pfeerick, 2023-07-03:
It explains why there could be a problem for 2.9 onwards, but not for 2.8.4. Something else must be going on there.

3djc, 2023-07-03:
> It explains why there could be a problem for 2.9 onwards, but not for 2.8.4. Something else must be going on there.

Indeed. My x7 unfortunalty isn't an S version

philmoz, 2023-07-03:
If I force IS_BLUETOOTH_CHIP_PRESENT to true in bluetooth_driver.h and build the X7 firmware then I get the bluetooth settings on my X7S.

If I then change the mode on the radio it finds a bluetooth address.

The Frsky app on iOS detects the radio when BT is set to telemetry (other than that the app is useless).

3djc, 2023-07-03:
@ParkerEde Could you try : https://github.com/EdgeTX/edgetx/tree/3djc/fix-x7s-bt-detection

ParkerEde, 2023-07-03:
@3djc Yeah, that works. Tested Trainer funktion with my X10S Express. Well done

ParkerEde, 2023-07-05:
@pfeerick can we have this fix in 2.9?

pfeerick, 2023-07-06:
Not until Raphael has reviewed the PR to see if this is the way we want to fix this going forward. If that isn't the case, this doesn't necessarily prevent it from going into 2.9 though as a temporary fix.

closed 2023-07-12 by pfeerick

Zwaubel, 2024-09-27:
Hi,
I'm currently facing the same issue on a Taranis QX7s running EdgeTX 2.10.4.
The Bluetooth menu entry in the hardware settings panel (where it used to be) is still absent.
Is this issue still unsolved?

pfeerick, 2024-09-30:
SYS => VERSION => Firmware options screen... does it list "bluetooth" as an option? If so, the code actually probes for a response from the bluetooth module, so either that is not responding, or isn't present... as it was confirmed to be functioning again in 2.9 (after being broken in 2.8).

Zwaubel, 2024-10-01:
Yes, it's there, and I can find my X7s as a Bluetooth device via a phone. 
I can even change the advertisement name via the desktop companion ap. 
It is just to present via the UI of the transmitter itself.

Zwaubel, 2024-10-12:
So... When using EdgeTX 2.9.4 the Bluetooth settings option is there. Everything works as expected. But when switching back to 2.10.4 that settings option is gone again. That to me indicates that on 2.10.4 something broke again for the x7s and its Bluetooth module.

Zwaubel, 2024-10-12:
I now went through 2.10.0, 2.10.1, 2.10.2 and 2.10.3 without success regarding Bluetooth on the x7s. Seems like something changed in the way 2.10.x probes and communicates with the Bluetooth module.

oilytin, 2025-02-18:
Same issue for me, bluetooth setting there in 2.9.4 but missing in 2.10.6

turibiusrozario, 2025-02-26:
For anyone in the future who stumbles on this like I did: I went to EdgeTX Companion, connected my transmitter and read settings, and then edited radio settings. In the radio settings, I could see the Bluetooth settings set as "OFF"; I switched it to "Trainer". From there, Bluetooth functionality was restored for trainer. This was with EdgeTX 2.10.5.

pfeerick, 2025-02-26:
Prior to that,  was there not a ```Bluetooth   --- ``` line present in SYS -> Hardware (just under the Internal RF settings)? You should have been able to select that line, and change it from the `---` (OFF) position to either Telemetry or Trainer.

PR #3754 (2023-07-12): Add X7S BT detection
This fixes #3752