EdgeTX/edgetx #3760: 2.9 Themes

tourerjim, 2023-07-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

TX16s MKII updated to 2.9 but my own Themes screen goes black and not able to read anything from the screen, went back to 2.8.4 and my Themes screens fine and with 2.8.3

Cant edit my Themes on 2.9, cant see 'Edit' or the drop down to edit.

### Expected Behavior

I assume I just  copy over my sc card content and my theme.yml works

### Steps To Reproduce

If I change from EdgeTX theme to my them screen goes black 

### Version

2.8.4

### Transmitter

Radiomaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### OS Version

none

philmoz, 2023-07-04:
Please post your theme.yml file here so we can see what is wrong. You will need to rename it to theme.yml.txt.

tourerjim, 2023-07-05:
summary:
  name: FM2M_005
  author: Jim
  info: theme 005

colors:
  PRIMARY1:   0x000000
  PRIMARY2:   0xF0FCF0
  PRIMARY3:   0xF80000
  SECONDARY1: 0x000400
  SECONDARY2: 0xA8ACA8
  SECONDARY3: 0xB8CCE0
  FOCUS:      0x407400
  EDIT:       0x4890E0
  ACTIVE:     0xF8E418
  WARNING:    0xB0B0B0
  DISABLED:   0x808080

pfeerick, 2023-07-05:
You'll need to upload this on Github - email attachments are ignored.

tourerjim, 2023-07-05:
[theme.yml.txt..txt](https://github.com/EdgeTX/edgetx/files/11954226/theme.yml.txt.txt)

philmoz, 2023-07-05:
Open the file in a text editor and remove the blank like before the 'colors:' section.

tourerjim, 2023-07-05:
Thank you,  removing blank lines my themes are now working with 2.9

tourerjim, 2023-07-05:
Unfortunately there is still bit of an issue, to get my theme working I need to edit within your theme, correcting the spacing only worked when I edited within your theme ie FM2M_Concrete, if I transfer my theme from 2.8.4 or write my own it will not work and looking at the two pictures may help.
picture 1 is your .yml file (FM2M_Concrete) and file shows thunbnail blank, my edit picture 2 saved as .yml shows text lines on thumbnail.
[image]
[image]

closed 2023-07-05 by tourerjim

philmoz, 2023-07-05:
Check that Windows is not hiding a .txt extension on the edited file.

tourerjim, 2023-07-05:
Could you show example, I save in Notepad as theme.yml

pfeerick, 2023-07-05:
In your own screenshot, you can't see an extension for the logo file, so the extension for your theme file may or may not be `.yml`. You need to go into the file explorer options, and untick the "hide extensions for known file types" option.
[image]

Alternately, when you save it in notepad, make sure you change the save as type from "text document (*.txt) to "All files (*.*)", otherwise notepad will always append .txt to the filename whatever you give it ;)
[image]

philmoz, 2023-07-05:
Select the 'Options' dialog under the View menu in Windows Explorer.
Uncheck the 'Hide extensions for known file types' option in the View tab.

[image]

tourerjim, 2023-07-05:
Thank you, I'll try that.

tourerjim, 2023-07-05:
Alls good now, on 2.8.4 or lower I didn't have issues with image naming 'logo' and saving or .yml files in Notepad, but something has changed in 2.9 saving logo is now saved as logo.png & theme.yml as your 'image file options' theme.yml is now saved as 'all file types.

pfeerick, 2023-07-06:
Neither the extension for the logo file (`.png) or the theme file (`.yml`) have changed in 2.9. 

Have a look at the name of the option in your file browser we directed you to, it isn't just "hide file extensions", it's "hide **known** file extensions". In the first screenshot you showed in https://github.com/EdgeTX/edgetx/issues/3760#issuecomment-1621361234, your windows install doesn't recognise the file type, so shows a generic icon. In the second screenshot, it shows a different icon, hence the file extension (which you can't see, otherwise you'd also be able to see the extension for the logo file next to it was .png), so really that file was called `theme.yml.txt`, which is invalid. If you check on the radio using the SD browser, you'll see this also. You really need to turn that option off so that it isn't misleading you.

tourerjim, 2023-07-06:
Not sure if it's to do with Windows which done update on my laptop couple days prior to me updating to 2.9.
For me to edit theme.yml to work with 2.9 this is the file thumbnail that shows in Themes.

[image]
If I drop this file to here I get message (we don't support that file) but and strange thing is that same file works with 2.9 Themes.

pfeerick, 2023-07-06:
More importantly, you can now see the file extensions for both, so know they are both correct. If you wanted to upload it to github, you would have to do what philmoz said back at https://github.com/EdgeTX/edgetx/issues/3760#issuecomment-1620790138, which was 

> You will need to rename it to theme.yml.txt.

in other words, change the file extension so it ends with `.txt`, which is what GitHub said was one of the supported file extensions when you tried to upload the file. I think this is how you got yourself confused before, as when you changed the filename from `theme.yml` to `theme.yml.txt`, it became a "recognised file type", and thus the extension was hidden, meaning you could only see `theme.yml`. It's only when you turned that extension hiding nonsense off that you can see what the true file extension was. 

Anyway, the important thing here is there was a change in how EdgeTX 2.9 reads/writes `theme.yml` files which you've helped us identify, and it's just a matter of ensuring there are no blank lines in the file to confuse the firmware.

tourerjim, 2023-07-06:
Your helps are well appreciated, an old boy trying to keep up with all this tech and am really pleased I've got all my own themes back now working.
Thank you.

PR #3762 (2023-09-07): fix(color): Fix handling of blank lines in YAML files.
Fixes #3760 

Better handling of blank lines in YAML files in the case where the blank line was at the start of the parse buffer after a newline ended the previous buffer.