EdgeTX/edgetx #3771: SE and SF not recognize

Djahmo, 2023-07-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Simulator

### Current Behavior

In joystick mode on Boxer ans Tango 2, the SE and SF joystick are not recognize by the computer ans also the sim

### Expected Behavior

Make that switch work on sim

### Steps To Reproduce

1. Connect radio on pc (bt or usb)
2. Try to push on SE or SF

### Version

2.8.0

### Transmitter

Radiomaster Boxer

### Operating System (OS)

Windows

### OS Version

Windows 11

3djc, 2023-07-06:
You need to configure your radio to turn those into channels. Not a bug

pfeerick, 2023-07-06:
In addition to what JC asked, How are your mixes set up? Are SE/SF assigned?

Djahmo, 2023-07-07:
> Are you talking about quad sim or companion sim ?

Quad Sim like : TrypFPV, Uncrached, LiftOff etc..

Djahmo, 2023-07-07:
> In addition to what JC asked, How are your mixes set up? Are SE/SF assigned?
I just connect my radio by USB (bt dont work for me, not detected on W11) and i select Joystick mod on Boxer's screen. 
When i push normal switch that work but the switch on  top side dont is not recognized. 
I flash my radio with ExpressLRS app

3djc, 2023-07-07:
Need appropriate setup: 

https://www.youtube.com/watch?v=OuLqiI5OwmQ&ab_channel=BillClark

closed 2023-07-07 by 3djc

phileville, 2023-07-07:
In order for the switches to work in the sim, they need to be added to the mixer page on your radio like you would when you want to use it for one of your real models.  USB mode sends the configured radio outputs to USB, so if your switches are not setup in a on the radio in a mix, then they will not be in an output and therefore not recognized in the USB.