EdgeTX/edgetx #3776: CPN (possibly radios too): b&w radio Simu crashes on scrolling through telemetry protocol option

mha1, 2023-07-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

b&w radio Simu crashes on scrolling through telemetry protocol option

### Expected Behavior

Don't crash. But the relevant question is what good does this option do bearing in mind color radios don't have this option and the option list in CPN has only two entries while the b&w radios have far more list entries

### Steps To Reproduce

1. start 2.8 Companion
2. create new radio profile, e.g. x7/x7s access and select it
3. create new model (file/new, +model,  cancel wizard)
4. double click on MODEL01
5. in tab SETUP select PPM as External Radio Module
6. click on Simulate
7. navigate to TELEMETRY settings
8. observe Type being set to FrSky S.PORT
9. click on Enter
10. use mouse wheel to browse through the list entries
11. observe CPN crashing on scrolling past Multi Telemetry

### Version

2.8.3

### Transmitter

FrSky Q X7 / Q X7S (ACCESS)

### Operating System (OS)

Windows

### OS Version

W10

### Anything else?

Notes:
- this might crash b&w radio firmware too
- same crash occurs in 2.9RC1 and 2.10 Simu versions

ParkerEde, 2023-07-09:
here a video

https://github.com/EdgeTX/edgetx/assets/15068446/b0ba39f4-078f-4008-8883-204302f6680c

mha1, 2023-07-09:
@ParkerEde Thank you very much, at least it's not crashing like the simulator

@pfeerick The missing line and simu crash is due to a mismatch (most likely since a very long time) between
`#define TR_TELEMETRY_PROTOCOLS         "FrSky S.PORT","FrSky D","FrSky D (cable)","TBS Crossfire","Spektrum","AFHDS2A IBUS","Multi Telemetry"` and
```
enum TelemetryProtocol
{
  PROTOCOL_TELEMETRY_FIRST,
  PROTOCOL_TELEMETRY_FRSKY_SPORT = PROTOCOL_TELEMETRY_FIRST,
  PROTOCOL_TELEMETRY_FRSKY_D,
  PROTOCOL_TELEMETRY_FRSKY_D_SECONDARY,
  PROTOCOL_TELEMETRY_CROSSFIRE,
  PROTOCOL_TELEMETRY_SPEKTRUM,
  PROTOCOL_TELEMETRY_FLYSKY_IBUS,
  PROTOCOL_TELEMETRY_HITEC,
  PROTOCOL_TELEMETRY_HOTT,
  PROTOCOL_TELEMETRY_MLINK,
  PROTOCOL_TELEMETRY_MULTIMODULE,
  PROTOCOL_TELEMETRY_AFHDS3,
  PROTOCOL_TELEMETRY_GHOST,
  PROTOCOL_TELEMETRY_FLYSKY_NV14,
  PROTOCOL_TELEMETRY_DSMP,
  PROTOCOL_TELEMETRY_LAST=PROTOCOL_TELEMETRY_DSMP,
  PROTOCOL_TELEMETRY_LUA
};
```

Current state of affairs:
- the "Type" edit field is not present on color radios in 2.8, 2.9 and 2.10 and probably also going back a long time, b&w only (very inconsistent) with PPM selected as external module
- 2.9 and 2.10 radio firmware does not have any meaningful functionality behind the data entered due to the module/serial API rework
- 2.8, 2.9 and 2.10 simulator crashes when browsing past the Multi Telemetry list entry 
- 2.10 will have the "Type" entry removed as per our discussion

Please advise on how to proceed. I think there are only bad options:
- leave 2.8 and 2.9 as they are. It is only me complaining about companion having only two protocols selectable, simulator crashing and radio fw having an empty list entry
- fix the 2.8 and 2.9 UI portion knowing 2.9 doesn't have functionality behind it
- fix 2.8 and leave 2.9 for probably zero complaints and knowing 2.9 will be imperfect (who would have thought)

To be honest I'd leave 2.8 and 2.9 as they are as obviously no one noticed anything in 2.8. Nothing bad happens in 2.9 and 2.10 will have this mess removed. But I'm open to all proposal.

mha1, 2023-07-16:
You adequately described my life.

My opinion:
2.8: no complaints about CPN crashing or missing functionality  - leave as is
2.9: fixing the UI issue doesn't bring back functionality. Leave as is, maybe fix CPN crashing but there were no complaints in 2.8 (and 2.9rc1 so far)
2.10: The chop shop is open: #3782. Bring back functions on demand only.

mha1, 2023-07-16:
btw: I'd rather disable CLI selection on AUX1/2 in 2.9 as there are still loose ends from #3055

closed 2023-09-07 by pfeerick