EdgeTX/edgetx #3794: Trim values different on Companion and Radio

Shonkonray, 2023-07-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The trim values shown on the transmitter display (or companion simulator) are different than what is shown in companion under flight modes. The values shown under flight mode are the same as in the model.yaml.
I converted from OpenTx and the trim values in the yaml file and companion match with the ones i had in OpenTx.
So i think something is wrong with how they get displayed, it seems that there is some kind of conversion factor included.

### Expected Behavior

Values on display and companion/yaml file should be the same

### Steps To Reproduce

1) Select a model in companion
2) Open flight modes
3) Add some trims (i tested 5,10,15,20)
4) Press simulate (with the values mentioned in step three, the simulation was showing 1,2,3,4)

### Version

2.8.4

### Transmitter

FrSky X10 / X10S (ACCST)

### Operating System (OS)

Windows

### Anything else?

[image]

pfeerick, 2023-07-13:
Also 2.9-RC1 ... The trim value specified was 10, as shown in the tooltip. I got 8 for 10, and 16 for 20... trim steps had no effect, so definelty seems a scaling issue. Limitied to colorlcd - B&W 128, 212 (TX12, X9D+2019) don't exhibit this. 

[image]

philmoz, 2023-07-13:
The radio trim display is always % (-100 to +100).
The companion / yaml value is raw trim value -128 to 128, or -512 to +512 for extended trims.

pfeerick, 2023-07-13:
That is only the case for colorlcd - B&W shows the value as encoded - i.e. 128 or 512. So the question is then do we change companion AND B&W to match, or make colorlcd behave the same as the others (and also how it used to display with OTX)?

philmoz, 2023-07-13:
Changing the color LCD display logic would be simplest; but was there a reason for the % display?

Perhaps it should be user selectable.

3djc, 2023-07-13:
Bw radio likely don't have the luxury to display a 0,1% scale

Shonkonray, 2023-07-13:
That i dont know, but than i would at least opt for same values between TX and Companion. And the 0.1% as a bonus stage if possible.

pfeerick, 2023-07-14:
It looks like it changed to a % early in the OTX 2.4 days, and hasn't been touched since. IMO the value itself is of no importance, as long as it is comparable. Raw values make sense if you don't want to mess with fractional numbers. % can look nice if it as the measurement percentage of trim. I would opt for bring colorlcd back into link with Companion & B&W UI. As a further/later extension, consider an option for showing the percentage. I'm not completely sold on that being needed, seeing you can see both the uS and % values in the channel monitor (so why clutter the main view more), but it's still worth discussing ;)