EdgeTX/edgetx #3795: Change Sticky function or add a second one

Shonkonray, 2023-07-12:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I would like to have a different behaviour of the logical switch function "Sticky".

As it is now the Sticky gets active when V1 changes from FALSE to TRUE regardless of the condition of V2. And it resets when V2 changes from FALSE to TRUE regardless of V1.

This, to me, is annoying because without the use of a second (or more) logical switch i cannot program the "Sticky" in a "safe" way or in other words, achive that the Sticky only gets activated in only one way and always in the same way and order.
That means, with the Sticky alone, i cannot guarantee that V1 can only activate the Sticky AFTER V2 has been set to the right position.

An example would be, that i want to activate a calibration mode on my models. But i dont want to do this by accident or in a random or "unpredictable" way. And i think the Sticky is unpredictable, especially if combined with its inate AND switch, there is no way to know whats going to happen after the AND gets TRUE. V1 and V2 could have been switched around randomly while the AND switch is FALSE, thus the state of the Sticky part of the logical switch can be anything and is unknown to the user.

### Describe the solution you'd like

What i want to have is a RESET dominat Switch (Sticky), which means that as long as V2 is TRUE it doesnt matter what happens on V1. Than after V2 changes to FALSE it needs a rising trigger (change from FALSE to TRUE) on V1 to be set and it will remain set as long as V2 stays FALSE. As soon as V2 changes to FALSE the Sticky is reset and FALSE.

This would also aliviate the intransparency of the state of the Sticky part while combined with its AND switch.

So the only thing that would need to be changed in the current version is to remove the hidden/underlying rising trigger/edge on V2.

### Describe alternatives you've considered

The alternatives i have to use now are more lines of logical switches to guarante my desired activation order of switches and maintain a clear state of the Sticky.

### Additional context

In this example L09 is what i wanted to do, but L12 and L13 is what i needed to do with current Sticky.

I feel that the Sticky would be easier to understand and use with this change.
In addition i believe that the use case is probably more often as RESET dominat with a clear way what has to happen first to activate/deactivate the Sticky.
[image]

pfeerick, 2023-07-13:
I'd have to think about it more, but I think your proposal actually breaks the inherent functionality of a sticky / set-reset switch. i.e. would this still function as a toggle (which is probably the most common use for it)?

[image]

Shonkonray, 2023-07-13:
You are right, i didnt think of that use case, nor would i have figured out that this would work. The impression of the underlying logic of the Sticky i had was entirely wrong and its very different programmed 😕
But yes, my suggestion would break this logic, which would be bad.

So, if possible, i would like to see an additional safety function as described above. 😄

pfeerick, 2023-07-14:
I think there is a proposal for something like this already, so you may very well see something like that soon. 

This is some logic I use, might be another way to look at it. I use this for arming/disarming, conditionally. Still has one more stage to be added, a hard-cut manual override, but it's been working quite well for me. 

[image]

In a nutshell 
- L01 gives me a pulse when SF goes down [arm switch **must** be going from up to down]
- L02 checks if the throttle is down (with a little wiggle room) [throttle down]
- L03 combines those two (so true if SF is down, and the throttle is down) [safe to arm]
- L04 combines throttle down and SF up [disarm]
- L05 latches if L03 true, resets if L04 true [system armed]

The purpose of L01 is to make it so if the throttle is up at the time I try to arm (SF down), I must lower the throttle, AND reset the switch (back to SF up) - basically start again - before it will arm. 

Bonus extras:
- L06 enables if the throttle is not down (!L02) while the system is armed [in flight]
- L07 enables if SF goes down while in flight [disarm in flight detected]
- L10 checks if system not armed, and no telemetry
- L11 checks if system is armed, and telemetry present

L06 and L07 allow me to detect an attempt to disarm in flight, and L07 triggers a beep to let me know if I pull the throttle down while in the current state, it's bye bye quad. 

L10 is used to reset my flight timers when telemetry lost and system is disarmed. Basically when I pull the battery, it auto resets the timer. L11 is used to run logging only if I have telemetry and am armed. 

Am I reading your intent correctly there that you want to use SC- (mid) for calibration, but only if the 6POS is not in the 6th position?

Shonkonray, 2023-07-14:
Many parts of your throttle start are similar to what i am using, thanks for the write up. 👍 

My intent is that I want to activate calibration with SG- **ONLY** when 6Pos is **IN** the 6th position. The same way you are forcing yourself to move throttle stick BEFORE you can arm. Its the same principle.
And to deactivate calibration mode i need to change the position of my 6pos switch to anything other than the 6th pos.

With my current setup L12 and L13 i am achieving this, but L09 is what i wanted it to look and work like, because its just easier, safer and better for such pruposes. Please see the screenshot above for the logic.

I could use any other switch instead of the 6Pos, but i opted for this one, because its very unlikely that i will activate or use this one during flight. All other switches are more prone to unwanted uses or are to valuable as an actual switch in flight.

Also there is [this](https://github.com/EdgeTX/edgetx/discussions/2587) discussion on the Sticky switch and the problem he is describing would be solved with my idea of a Reset dominant switch. 😄

Would be great if you are right and there is something like this coming soon.

3djc, 2023-07-14:
If you use a cut switch (which you should) you can also ensure the cut switch is in cut position before enabling calibration

Shonkonray, 2023-07-14:
> If you use a cut switch (which you should) you can also ensure the cut switch is in cut position before enabling calibration

Thank you for the heads up. I have a cut switch in use, but i have not decided on how i want it to interact with calib and what exactly should be possible inside this calib mode. But that would be another discussion for another place.