EdgeTX/edgetx #3806: X9d 2019 internal RF

lhh6688, 2023-07-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

EdgeTX X9d 2019 internal RF fail that is ACCeSS not XJT
OpenTX is fine

[image]

[image]

### Expected Behavior

X9d 2019 internal RF

### Steps To Reproduce

X9d 2019 internal RF

### Version

2.9.0

### Transmitter

FrSky X9D+2019

### Operating System (OS)

Windows

pfeerick, 2023-07-16:
Can you show the SYS -> VERSION page please... for the X9D+2019, you can't change the internal RF type setting (which could have explained why it said XJT instead of ISRM), so it is almost like you have loaded the X9D+ firmware, instead of the X9D+2019 firmware. 

[image]
[image]

lhh6688, 2023-07-17:
Firmware is fine
[image]

3djc, 2023-07-17:
Could you show this screen

[image]

If you had loaded a wrong one and set XJT, it could still be there

lhh6688, 2023-07-18:
> Could you show this screen
> 
> [image]
> 
> If you had loaded a wrong one and set XJT, it could still be there

[image]
No XJT

pfeerick, 2023-07-18:
Given that the option to change the internal RF is not present on the
X9D+2019, all I can think off is perhaps that option was set by an
incorrect firmware flash, and now you'll need to change it manually.

Open up the RADIO/radio.yml file in a text editor (i.e. Notepad), and make
sure internalModule is as follows

internalModule: TYPE_ISRM_PXX2

You'll also need to change the parameter on the second line, manuallyEdited
from 0 to 1, so that the change to the file is accepted. You should be
right from then onwards. You 'may' find you need to turn the internal RF
setting off and on again in the model settings if it's still saying XJT
there.

lhh6688, 2023-07-18:
> Given that the option to change the internal RF is not present on the X9D+2019, all I can think off is perhaps that option was set by an incorrect firmware flash, and now you'll need to change it [...]
you are right !
I change it that make X9d 2019 fine
Thank you ,

pfeerick, 2023-07-19:
Glad that fixed it for you :)

closed 2023-07-19 by pfeerick