EdgeTX/edgetx #3812: No BST telemetry on crossfire Nano diversity Rx adapter

ror1948, 2023-07-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

My crossfire  BST telemetry(4 entries) Nano Rx worked in the past and I have been loading the nightly builds on my TX16s.
the other normal telemetry from a nano Rx is working fine
currently loaded is 2.10.0 B0e64c8d
I tried 2.8.4 but still BST did not work.
also tried tx16s290rc1 but same problem.
crossfire 6.19
missing RxBt, curr, capa, Bat% from a current sensor on BST line
I am not sure it is the RM firmware but that is what I have been changing lately.
Maybe you can suggest another possible problem.
attached are 2 models that had the BST working with 2 different hardware Rx
thanks

### Expected Behavior

those extra 4 BST telemetry lines to work.

### Steps To Reproduce

RM Tx on
Rx on
all telemetry entries should be being updated.

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### Anything else?

currently loaded is 2.10.0 B0e64c8d
date 2023-07-11
time 2;46;57

ror1948, 2023-07-16:
here is setups
[crossfireBST.zip](https://github.com/EdgeTX/edgetx/files/12064543/crossfireBST.zip)

ror1948, 2023-07-16:
this is the Rx adapter
https://www.team-blacksheep.com/products/prod:nano_divrx_adptr

this is the BST current sensor
https://www.greathobbies.com/productinfo/?prod_id=TBSCSPRO100A

ror1948, 2023-07-18:
It is a problem with the crossfire firmware.
It works ok with CRSF 6.07 but not with 6.19
Now what are the chances that TBS will fix the firmware?

need CRSF 6.07 for the TBS CROSSFIRE NANO DIVERSITY RX ADAPTER  and BST telemetry to work
need CRSF 6.19 for the Matek CRSF-PWM-V10  telemetry to work.

All parts of CRSF system must have same version

now what?

pfeerick, 2023-07-18:
> Now what are the chances that TBS will fix the firmware?

Better than they were a few months ago... it appears TBS is coming out of the slump it's had due to covid/restrictions, staff shortages and internal disputes.

ror1948, 2023-07-18:
yes I saw that.
Most likely will not wait for the fix.
Not really that high a cost.
This crossfire and ELRS is great but they lack telemetry unless you pair them with the Matek PWM adapters to give some telemetry.
I am a wing flyer rather than a quad flyer.

thanks

pfeerick, 2023-07-18:
Yeah, external telemetry for ELRS is still early days... but I believe plans are in the works for it. 

It may still be good to directly contact TBS support so they are aware of this bug (as a customer, I consider this a bug being forced to stick with an older firmware version for it to work) so they at least can't say they aren't aware of it. Given it looks to be a discontinued line, I wouldn't hold my breath though.

ror1948, 2023-07-18:
Yes I am talking to their support already and have told them it is their firmware....waiting for reply
now do they believe me..

ror1948, 2024-06-08:
still no fix in crossfire 6.31 so I guess not going to happen