EdgeTX/edgetx #3818: Variometer tone doesn't work with ranger nano and elrs receiver

Le0Michine, 2023-07-17:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When I enable variometer tone nothing happens. It doesn't emit neutral tone and doesn't react on VSpd value change.
Other sounds are working just fine so it isn't an issue with radio being muted.

### Expected Behavior

A neutral tone when model isn't flying, different tone when model ascends/descends in the air

### Steps To Reproduce

My setup: 

Radio: radiomaster zorro (multi internal module) + ranger nano with ELRS. 
Receiver: radiomaster ER6GV ELRS receiver
Model config: using external module with CRSF protocol, telemetry working including VSpd and Alt values

Vario related config:
1. For model in Telemetry section set Vario/Source to "VSpd"
2. Set Vario/Center to -0.5 0.5 Tone

### Version

2.8.4

### Transmitter

Radiomaster Zorro

### Anything else?

I saw an old issue with the same symptoms, seems like it was related to usage of external Tx module https://github.com/EdgeTX/edgetx/issues/2546
Unfortunately I can't test this with internal module as I don't have any non ELRS receivers with telemetry. I haven't tested it with 2.9 as it hasn't been released yet and it doesn't seem like there is anything related to my issue there.

Le0Michine, 2023-07-17:
Attaching radio and relevant model configurations (edited out owner id)

[radio_model.zip](https://github.com/EdgeTX/edgetx/files/12074223/radio_model.zip)

pfeerick, 2023-07-18:
Looks like you don't have a "Vario" special function to actually turn on/off the vario? Both internal and external ELRS are working for me with Zorro and 2.8.1

Le0Michine, 2023-07-18:
can you clarify what special function you talking about? From the documentation here https://edgetx.gitbook.io/edgetx-user-manual/b-and-w-radios/model-select/telemetry it looks like there shouldn't be any need for extra configuration to be done besides setup in model -> telemetry -> vario section

Le0Michine, 2023-07-18:
found it, thanks a lot for the tip, seems like the documentation could be improved on that topic

after I added special function "ON - Vario" it started beeping as expected

closed 2023-07-18 by Le0Michine