EdgeTX/edgetx #3829: (feat) Flysky Gimbal Support for T20

richardclli, 2023-07-19:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Flysky gimbals can be installed properly in T20, now is time to enable the firmware support.

### Describe the solution you'd like

Flysky gimbal support for T20.

raphaelcoeffic, 2023-07-20:
On `main` this should be only about adding a few defines to `hal.h` and `FLYSKY_GIMBAL` to `CMakeLists.txt`.

richardclli, 2023-07-23:
Yes, I know.  Trying to.work this out.

rotorman, 2023-07-24:
You need to define also FLYSKY_HALL_* as e.g. here: https://github.com/EdgeTX/edgetx/blob/5386fe92adf03f8990cfe13e3714f4ffb217eb3a/radio/src/targets/taranis/hal.h#L2706-L2723

PR #3854 (2023-07-31): feat: Flysky gimbal support for T20.

Fixes #3829

Summary of changes:
closed 2023-07-31 by pfeerick