EdgeTX/edgetx #3830: radiomaster zorro spash screen bug

LexFPV, 2023-07-19:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

i just updated to edgetx 2.9.0 rc1 and im having an issue with the splash screen.

when i power on and the edgetx logo is shown it looks like it tries to put a black background onto the screen while its white
(see screenshots)

this did not happen when i was on 2.8.4.!

[image]

### Expected Behavior

that it boots up normally without the black backdrop.

### Steps To Reproduce

power radio on and look at the splash screen

### Version

2.9.0

### Transmitter

Radiomaster Zorro

### Operating System (OS)

Windows

### OS Version

Windows 11, version 22H2

pfeerick, 2023-07-19:
Check your displays contrast setting, it's probably a fraction too high.

3djc, 2023-07-19:
Pfeerick is completely right, but I must add that there is no settings that will make it right for all screen. If it is perfect for this one, it will worse in others. This is not a software issue, it is down to the lcd panel power driver circuit (and there I must also add that is is not a Radiomaster issue, all similar radio, including FrSky x7 suffer the very same issue)

LexFPV, 2023-07-19:
> Pfeerick is completely right, but I must add that there is no settings that will make it right for all screen. If it is perfect for this one, it will be worse in others. This is not a software issue, it is down to the lcd panel built in power driver circuit (and there I must also add that is is not a Radiomaster issue, all similar radio, including FrSky x7 suffer the very same issue)

yeah okay. the thing is that even on contrast level 10 it is still there. and it wasnt there before i updated to 2.9.0(was on 2.8.4 before)

3djc, 2023-07-19:
In addition I checked lcd_driver_spi.cpp where this is happening, there no change between 2.8 and 2.9 branches

LexFPV, 2023-07-19:
> In addition I checked lcd_driver_spi.cpp where this is happening, there no change between 2.8 and 2.9 branches

huh.. so this is just the lcd?

LexFPV, 2023-07-19:
> No you are right, there seems to be a difference, trying to narrow it down

okay.!

3djc, 2023-07-19:
Got it, submitting fix PR

3djc, 2023-07-19:
As a side note @pfeerick , boot logo is displayed before radio data is loaded so using LCD_CONTRAST_DEFAULT

pfeerick, 2023-07-20:
No, that's not quite true... otherwise, the radio wouldn't know to apply the startup delay, or how long to show the logo (if at all). There is a pre-load of the settings to get those values, before choosing to run the animation / show splash screen. 

https://github.com/EdgeTX/edgetx/blob/2ab21da86622d08a917af99b58ea8b84664d22f1/radio/src/opentx.cpp#L1445-L1467

I think the issue now is that lcdSetContrast isn't called again after board initalisation using `LCD_CONTRAST_DEFAULT` until later in startup. 

https://github.com/EdgeTX/edgetx/blob/2ab21da86622d08a917af99b58ea8b84664d22f1/radio/src/opentx.cpp#L1591-L1593

PR #3831 (2023-07-23): fix: revert Zorro default contrast
Revert a change introduced in https://github.com/EdgeTX/edgetx/pull/3575

With settings introduced in 3575, author custom splash with a lot of black pixels looks better, but default EdgeTX and many other look worse. Suggesting reverting to initial value

This fixes #3830

Recommend applying to 2.9
closed 2023-07-23 by pfeerick