EdgeTX/edgetx #3833: Bootloader in Branch 2.9 defective

ParkerEde, 2023-07-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I have now built and tested the latest Branch 2.9 for the Horus X10S Express. After flashing the firmware, I pressed the RTN button as normal so that the Horus would start the updated firmware from the boot loader. Unfortunately, the emergency mode came up instead. Then I switched off the Horus completely and restarted it. This worked. Now I tried to flash the bootloader. During this, the Horus suddenly went completely off and could no longer be switched on. I could only revive the Horus via USB with the Zadig.

### Expected Behavior

Radio should start from Bootloader

### Steps To Reproduce

flash the 2.9 firmware
press the RTN button
emergency mode will come up
power off  and restarted it
Try to flash the bootloader (this will brick the bootloader)
revive the Horus via USB with the Zadig

### Version

2.9.0

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

### Operating System (OS)

Windows

raphaelcoeffic, 2023-07-20:
Did you try to flash the bootloader using DFU mode? Or did you only try to flash the bootloader from a running firmware?

ParkerEde, 2023-07-20:
Only from running Firmware

raphaelcoeffic, 2023-07-20:
> Only from running Firmware

Could you please try flashing with DFU mode? That should give us a better assessment of what it is actually not working (the bootloader itself, or the routine flashing from running firmware).

ParkerEde, 2023-07-20:
the flashing via DFU has worked. The Horus can also be started normally. Only if I then start in the bootloader and then exit, the emergency mode appears again

raphaelcoeffic, 2023-07-20:
Could you please try #3835?

ParkerEde, 2023-07-20:
I have already done that. I am in direct contact with mha1. But the PR does not change anything either.

ParkerEde, 2023-07-23:
I just built Branch 2.9 (67313cd) and tested it again. Now the problem does not seem to occur anymore.
Do we want to close here again or should it be investigated further?

pfeerick, 2023-07-24:
We're sure the issue was caused by the reverted commit so should be right
to close this now. Thanks for catching and reporting it ;)

ParkerEde, 2023-07-24:
Okay. I have now checked the main once without and once with PR #3837. Without PR #3837 it is OK, but as soon as I merge it, the error occurs.

mha1, 2023-07-25:
The bootloader writes the shutdown reason just before doing the system reset upon exiting the bootloader. Bootloaders prior to #3837 will not write a checksum which forces a checksum error at firmware startup (proving the mechanims works). The checksum error clears the RTC backup registers (default state according to STM32 datasheet) which is interpreted as "previous shutdown wasn't going well". Retesting with the bootloader also flashed to #3837 should result in a proper firmware startup after bootloader exit. Tests on TX16s worked. No EM after bootloader exit with #3837 bootloader.

This poses the question how should the backup registers be initialized after a RTC checksum error. Initializing RTC BKP0R to SOFTRESET_REQUEST would be an option but I still think clearing the registers and accepting an EM after a corrupt RTC memory is the safer option.

philmoz, 2023-07-25:
Since you can't force the bootloader to be updated before updating the firmware, this is going to result in a lot of people getting an EM error after upgrading firmware version.

This could result in a lot of support issues or a very negative reaction to EdgeTX. 
A firmware update that knowingly results in an EM startup should be avoided at all costs.

philmoz, 2023-07-25:
> kinda catch 22

Only if there is now way to differentiate between three possible states:
- No checksum (upgrade)
- Valid checksum
- Invalid checksum

mha1, 2023-07-25:
1. There may be a way (but I don't know it) to determine if the application was started from the bootloader. Not just a system reset, bootloader. The bootloader just perform a system reset (NVIC...)
2. no way because older bootloaders just don't generate and write it (catch 22 part).
3. same as 1. I think

mha1, 2023-07-25:
3. again: checksum is stored in BKP2R. Older bootloaders should have BKP2R set to 0 (default value that is never touched anywhere). So using checksum == 0 as invalid marker should work but it does compromise the checksum idea somewhat

ParkerEde, 2023-07-25:
I am reopening the issue because we still need to discuss it and we also need to make a decision on how to proceed.

pfeerick, 2023-07-26:
> 1. There may be a way (but I don't know it) to determine if the application was started from the bootloader. Not just a system reset, bootloader. The bootloader just perform a system reset (NVIC...)

Presently not AFAIK unless it's baked in the STM32 somewhere, but there should be a more portable method in the future... the conventional "write a flag somewhere in memory that the main app can read on startup" type. 

> 3 again: checksum is stored in BKP2R. Older bootloaders should have BKP2R set to 0 (default value that is never touched anywhere). So using checksum == 0 as invalid marker should work but it does compromise the checksum idea somewhat

Meaning we really only have two choices (which, given it's an data integrity check, just like with the radio.yml checksum, seems ok to me):
- valid EM data - > go into EM
- invalid/missing EM data -> startup normally

Or did I miss something there?

raphaelcoeffic, 2023-07-26:
> There may be a way (but I don't know it) to determine if the application was started from the bootloader. Not just a system reset, bootloader. The bootloader just perform a system reset (NVIC...)

The normal way of doing this is basically to declare a uninitialised section in RAM that is used to pass values across soft-reset. I'm working on it right now (see #3846). The idea is to have something that is more portable (as in: *usable on other MCUs*), simpler and better de-coupled / abstracted.

mha1, 2023-07-26:
Yes, I have done something similar to be able to run the bootloader from application code on STM32's that don't really support this. Magic word in RAM, changed linker script and startup asm.

But the catch 22 remains. You still need to update the bootloader prior to the application. DFU flashing with Cube Programmer solves that problem. Bootloader and application are written in a go. So buddy (never tried it) should do the same.

There might be one other possibility to maintain backwards compatibility with #3837. The bootloader only forwards BKP0R to the applications UNEXPECTED_SHUTDOWN() assessment. 
```
#if defined(RTC_BACKUP_RAM)
      rtcInit();
      RTC->BKP0R = SOFTRESET_REQUEST;
#endif
```
BPP1R and BKP2R should be still zero. Changing the simple checksum calc to result in a zero checksum for this combination should work.

raphaelcoeffic, 2023-07-26:
> DFU flashing

That would work with Buddy as well, no need to use explicitly the ST tools.

During our last meeting, @pfeerick suggested we might have a "hotfix" for 2.9 & `main`, and a longer term solution based on #3846 once ready and tested.

mha1, 2023-07-29:
I was able to find the reason for the EMs related to #3837. Bootloaders prior to #3837 only write RTC backup register 0 and don't write a checksum as expected by #3837. The assumption that backup registers 1 and 19 are zero is wrong. Those registers hold old values from previous startups/shutdown so the idea of changing the checksum calc to take the assumption in account didn't work.

Please see the (hack) solution in #3837 to solve the EM after exiting legacy bootloaders. It's working fine on my TX16s with the combination that gave me reproducible EMs upon exiting the bootloader - bootloader 2.8.1 and #3837 check firmware

@ParkerEde I'd very much appreciate you retesting #3837

ParkerEde, 2023-09-19:
I have now tested it starting from 2.8.5 and 2.9.0. I did not get an EM now

closed 2023-09-19 by raphaelcoeffic