EdgeTX/edgetx #3834: fw (installation) on x9d+ 2019 disables radio, bootloader menu no longer works

kamik2, 2023-07-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

7update the bootloader to 2.8.4 -> bootloader menu is no longer available
update main fw to 2.8.4 -> the radio no longer boots up

### Expected Behavior

.

### Steps To Reproduce

radion 2.7.1
- update bootloader by radio file system to 2.8.4
  - at this point, the radio no longer can enter the bootloader menu by trim 1+4 pressed while powering it up,
  - radio remains off
  - slight buzzing from the speaker noticeable
- update fw by buddy to 2.8.4
  - update installed without issues
  - radio no longer starts
  - bootloader menu is still not available

to get the radio back to live:
- flash by opentx companion with otx 2.3.15
  - radio boots again
- flash 2.84 etx by buddy agaib
- everything works now

### Version

2.8.4

### Transmitter

FrSky X9D+2019

### Operating System (OS)

Windows

### OS Version

win 10 home, 22H2, build 19045.3208

pfeerick, 2023-07-20:
Would the radio have had 2.7.1 bootloader at the time of the first upgrade?

kamik2, 2023-07-21:
yes,  the bl was on 2.7.1 before i updated it to 2.8.4

pfeerick, 2024-07-06:
Is it possible the wrong X9D firmware was flashed here the first time? As I've done this myself, and have had similar behaviour? Has this happened again in 2.9 or later?