EdgeTX/edgetx #3836: B&W 128x64 telemetry screen header is too crowded

lightbulbjim, 2023-07-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

EdgeTX 2.9-RC1 added the timer 1 value and time of day to the telemetry screen header. Unfortunately, once timer 1 reaches a sufficient value they collide.

[image]

### Expected Behavior

Probably the simplest option is to remove timer 1 from the header, leaving just the model name, TX voltage and time of day. Timer values can still be added to the telemetry page fields if desired.

### Steps To Reproduce

1. Configure timer 1 as a count up timer.
2. Let sufficient time elapse.
3. Observe a telemetry screen.

### Version

2.9.0

### Transmitter

Radiomaster Zorro

philmoz, 2023-07-21:
Maybe only display the minutes value if >= 100 (no seconds display).

lightbulbjim, 2023-07-21:
Actually if we're keeping the timer value I'd say it needs to be HH:MM if MM>=60. Seeing 276 minutes (as in the example above) isn't very useful.

So we'd still be looking at four digits plus the colon.

My vote is to just remove the timer value altogether, but I'm keen to hear what other people think :).

3djc, 2023-07-22:
Timer has been added recently due to users request...

lightbulbjim, 2023-07-22:
Can we make an option to toggle between time of day and the timer? They just don’t both fit on the small B&W display.

3djc, 2023-07-22:
The issue comes from the very high timer time, I would suggest that if timer value is large (tbd), there timer is likely not relevant to THIS flight and therefore should not be displayed

lightbulbjim, 2023-07-22:
Yes, that makes sense. For thermal gliders (like in this example) I usually use timer 1 as a session timer (for battery logging) and timer 2 as a flight timer. So during a day’s flying timer 1 can accumulate a few hours.

phileville, 2023-07-22:
I totally agree, the header is too crowded.  In my opinion the timer should be removed.  The timer can be configured if it is desired in one of the telemetry blocks.

pfeerick, 2023-07-23:
Voltage and timer were actually always there... but due to a bug, the timer was not visible... and then the time was added in also (both the fix to the timer, and the addition of time to the top bar were done in https://github.com/EdgeTX/edgetx/pull/3306 - looks like an undiscovered OTX bug got squashed there 🤪) 

Having said that, I agree with phileville... the timer can go... it's never been visible for all of ETX, and don't know how much of OTX (on the 128x64 displays)... so no-one will miss it.

3djc, 2023-07-25:
One could argue that vbat is kinda pointless there, so what could be done is display vbat if no timer is defined, or first timer if defined

lightbulbjim, 2023-07-25:
Everybody is going to have a preference for what to display. Personally I'd rather have vbat than the timer, but I don't really mind too much.

What I would say is avoid having it dynamic. Either always display one particular value, or make it an option in model settings.

philmoz, 2023-07-25:
Another option - alternate vbat and timer display in the same location.

[image]
[image]

lightbulbjim, 2023-07-25:
I strongly dislike auto cycling fields.

pfeerick, 2023-07-25:
Indeed - especially when the idea is that data should be there for you to read at a glance - not to wait for it to appear/cycle through. Without blowing this out to being yet another customisable option (for now), the aspect to me is what info would be important *across* screens, since you can have Timer1, Time and Batt as screen elements anyway. 

So personally, I would do Batt, and Timer1 if defined, else Time. With Batt on the far right, Timer to the left of it, always HH:MM:SS if timer, and HH:MM if time (to be able to differentiate between timer and time at a glance). On the premise that if you have a timer defined, that is more important than the time, and that the handset battery voltage is just as important if you only have the telemetry screen up!  

Also, in my (likely biased) opinion, persistent timers should not be in the T1 slot... I always use T3 for that :zany_face:

lightbulbjim, 2023-07-26:
> So personally, I would do Batt, and Timer1 if defined, else Time. With Batt on the far right, Timer to the left of it, always HH:MM:SS if timer, and HH:MM if time (to be able to differentiate [...]

That sounds fine. My personal preference is to keep things simple and predictable. At the end of the day I'll adapt to whatever is in the top bar; it's just broken right now.

> Also, in my (likely biased) opinion, persistent timers should not be in the T1 slot... I always use T3 for that 🤪

Maybe, but even better is for the user to be able to configure the timers as they see fit 🙂. Especially since the timer positions are hard coded on the main flight screen.

As far as how I ended up with my particular timer configuration... Once upon a time my fleet was 100% slope soarers, and I only had one timer configured because there was only one thing I cared about: battery runtime. I still have a bunch of slope soarers where this is the case.

3djc, 2023-07-26:
> Indeed - especially when the idea is that data should be there for you to read at a glance - not to wait for it to appear/cycle through. Without blowing this out to being yet another customisable [...]

Agree on the cycling thing, really not user friendly imho

Disagree on a couple of point. Time is important here, especially if you are flying glider (so long flight), time is needed if you have somewhere you need to be at specific time, so you know you need to leave airfield by ..

I don't think anyone would remeber when looking at a glance if it is differentiated by HH:MM:SS or HH:MM which one is which between time and a timer

I guess vbat must be something for ELRS flyers, because I could not care less about something during a flight

What we should chalenge then is the model name, whats the value of it on that screen ?

3djc, 2023-07-26:
but why would you want it there ? this is a while you fly screen

lightbulbjim, 2023-07-26:
I feel like I should have the model name on the screen somewhere, but that might just be habit. 

The reason for having it in the header is _because_ I don’t need to check it while flying. Anything I _do_ want to check while flying I would add as a field so it’s chunky and visible.

3djc, 2023-07-26:
If things are on the screen, it is because you might need it while flying. Model name isn't really fitting the bill

lightbulbjim, 2023-07-26:
Depends how you use it I guess. But that’s fine, as I said I don’t feel particularly strongly about what’s in the header. It’s just broken right now.

pfeerick, 2023-07-26:
> I guess vbat must be something for ELRS flyers, because I could not care
less about something during a flight

No is BATT, not vBatt, and has nothing to do with ELRS - it's the handset
voltage!!! Just like it is on OTX 🤭

Regarding the time... what, do people not wear watches, have phones, or car
clocks anymore? And, just sayin', it was never shown on 128x64 with OTX or
prior ETX versions, so it mustn't have been that important 🤣 But if you
insist on keeping that on-screen for those few glider pilots ... 🤪

But in all seriousness, removing the model name is probably the best
option, as that frees up the whole top bar for that data. Can always just
press RTN if you want to see the name, plus you need to anyway to do most
other tasks.

3djc, 2023-07-26:
Yeah I meant BATT, its just because only ELRS flyers care about it in flight, with 4in1 or xjt, it lasts so long on BW that it doesn't need to be displayed there, thats what I meant (and yes, it was in OTX, doesn't mean it was the best possible choice ! (yes, I said it, screenshot it :D :D ))

The time wasn't displayed, but it was asked somewhere I think, and definitively more usefull than model name on that screen imho

(and you should NOT have your smartphone in your pocket while flying)
(and watching your watch while holding your radio with two hands isn't that easy )

pfeerick, 2023-07-27:
> (yes, I said it, screenshot it :D :D ))

Screenshot what? 😇 😛 

Thinking about that a bit more... what do you think of making the model name conditional... timer if timer set, model name otherwise. It would want to be right aligned though, so that it doesn't runover into the void.

PR #3856 (2023-07-28): feat(128): Swap modelname for timer1 if defined
Resolves #3836

Swaps the model name for timer1 if it is defined. I was tempted to make it so if no RTC, the timer is drawn where the time would be, but that will then create a "if this handset, looks like this, if that handset, looks like that" documentation/tutorial issue. 

Timer 1 defined
[image]

No timer 1 (i.e. like before, minus timer)
[image]
closed 2023-07-28 by pfeerick