EdgeTX/edgetx #3841: Companion Soundpack Download, loads always the same file

Shonkonray, 2023-07-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

The soundpack download via companion downloads always the same .zip file, regardless which soundpack option is selcted.
Same behavior on V2.8.4 and V2.9.0 RC1

### Expected Behavior

The selected soundpack(s) (multi selection option is great, but doesnt work either), for example after selecting GB_RYAN, than only this should be downloaded. When selecting ryan and sara, both should be downloaded.

### Steps To Reproduce

Try and download different soundpacks via companion.

### Version

2.9.0

### Transmitter

FrSky X10 / X10S (ACCST)

### Operating System (OS)

Windows

elecpower, 2023-07-22:
Bug confirmed

pfeerick, 2023-07-23:
Oops... sorry! Fix one problem, create another! 🤦

elecpower, 2023-07-23:
I wasn't going to name names to protect the guilty.
The available releases are determined by the github release status and the companion release filter. Doesn't do a version check due to a number of factors.

pfeerick, 2023-07-23:
Jesus... need to have a word with the dude responsible for these namings ... 👀 
[image]

elecpower, 2023-07-23:
Lucky companion doesn't use descriptions but the release metadata to get version info

Shonkonray, 2023-07-23:
I am not sure if it is clear what i was talking about, or put another way, i am confused about what you are talking about. 😂
To add to the confusion i made a screenshot of what i was talking.
The selection popup "Choose language package" is downloading always the same file. In case that this was clear to you, sorry for double post. 😄 

[image]

elecpower, 2023-07-23:
@Shonkonray we know the issue from your perspective and it is caused by a 'behind the scenes' change. @pfeerick and I have been having a bit of devs banter.

PR #3848 (2023-09-07): fix(cpn): selection of sound packs in update process
Fixes #3841

Summary of changes:
- refactor asset filtering to not use sounds.json file but instead filter the release assets
- only 1 pack can be downloaded per run as all language variants share the same destination directory
- supporting functions
- housekeeping

[image]

Due to a change to the asset filter a release note is required as only 1 zip file will be found for each language with the old filter:
Steps:
1. Launch Companion
2. Select Settings
3. Select Update Settings tab
4. Click Sounds component Options button
5. Edit the Filter  [... cut]
closed 2023-09-07 by pfeerick