EdgeTX/edgetx #3842: Screen layouts can be overlapped

Shonkonray, 2023-07-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

If during widget setup the button "TELE" on the tx (or companion simulator) is pressed the layout selection screen pops up. There a new layout can be selected and if then the "setup widgets" option is pressed the old and the new layout are overlapped.
This can be repeated many times and widgets can be selected until at some point the system crashes. If one wants to abort setting up widgets, he has to cycle back with the return button as many instances as were prior "opened" with the tele button.

### Expected Behavior

I think best would be that the TELE button behaves like the RTN button, unless there is a good reason that it needs to be like this.

### Steps To Reproduce

1) Add telemetry screen
2) Press setup widgets
3) Press TELE button
4) Press setup widgets
5) Repeat as desired or until crash :)

### Version

2.9.0

### Transmitter

FrSky X10 / X10S (ACCST)

### Operating System (OS)

Windows

### Anything else?

Tested on 2.8.4 and 2.9.0_RC1, companion simulator behaves the same as TX.
[image]

Shonkonray, 2023-07-23:
Thank you for linking, and sorry that i have overlooked that there was already a pull request.

closed 2023-07-23 by Shonkonray

pfeerick, 2023-07-24:
I think the PR was in response to your report, so thanks for finding the
glitch in the machine ;)

PR #3845 (2023-07-25): fix(color): Change behaviour of TELE button on widget setup page to work like RTN button.
Prevents recursive loading of widget setup page.

Fixes #3842