EdgeTX/edgetx #3847: Flysky CAT-01 still reporting erroneous values in 2.9RC1

dand3, 2023-07-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Altitude/pressure values are way out by a very large amount
Issue exist in 2.83 (see  2.83 ->   PR#3583) but also 2.84 and now 2.9RC1
[image]

Hardware is TX16S - 
internal MPM  mm-stm-serial-aetr-v1.3.3.20  / AFHDS2
Receiver Flysky FS-IA6B  
Sensor is CAT-01 (* several sensors tested with same faulty results)

### Expected Behavior

ETX 2.82 with same CAT-01 sensor is reporting altitude/pressure correctly 
[image]

### Steps To Reproduce

Load ETX2.8.2 is OK
Load 2.8.3-2.8.4-2.9RC1  KO

### Version

2.9.0

### Transmitter

Radiomaster TX16S / TX16SMK2

### Operating System (OS)

Windows

dand3, 2023-07-23:
I volunteer for testing correcting software versiosn

pfeerick, 2023-07-24:
I'll trigger the rebuild of a pr that was intended to fix this and let you
know the id so can test it. It seemed mostly fine from when I tried it so
some additional testing would be appreciated.

dand3, 2023-07-24:
Good new. Awaiting test version..

pfeerick, 2023-07-25:
This is the PR I was thinking of : https://github.com/EdgeTX/edgetx/pull/3582. I've just rebased it, so it is basically what will be 2.9-RC2 + that change. Should be able to go to the Checks tab at the top of that PR, then click on "Run tests and package firmware" on the left side, and then click on "edgetx-firmware-merge" under the "Artifacts" heading to get the zip file with all the different firmware builds.

dand3, 2023-07-25:
TX16S loaded with ETX v2.10.0-PR3582 (ea7ce1fc)
MPM mm-stm-serial-aetr-v1.3.3.20
Temp 21.5 °C
Warmup time receiver & CAT01 sensor: 1 minute
Number of measurement for each sensor: 3 (spaced in time)
4# pieces CAT01 tested
'------------------------------------
Results are promising.
Reported AGL altitude for all 4 sensors: #1=116m - #2=128m - #3=109m - #4=130m
(One CAT01 sensor had a high t° drift and was took out of further test)
Offset seems high, but can easily be compensated in "alt" telemetry settings or better via "Auto Offset" (both tested).
A simple altitude test ( 1 floor) show's a 3.0 - 3.5m variation. Looks OK to me.
Could be confirmed with a higher delta ( high building , lift etc.) or flight with double log (for example with FrSky High precision Alt/vario for comparison). Will try this later on. 
Big thanks to all dev's for the hard work done (and going on) ☺

[image]
[image]

* pres diff is due to measurement at another time (pm).

CharlesGauci, 2023-07-25:
I had completely given up on this issue and decided to keep my NV14 on version 2.8.2 and shelved my EL18 for the time being. Telemetry on internal module of the EL18 is a mess on all versions and not just with the FS-CAT01. I am not aware of any recent updates since I made this decision last April. 

I cannot understand why Flysky decided to change the settings of the FS-CAT01 in version 2.8.3. If they stick to this decision, can anyone explain to us how to set up the telemetry screen?

I use voltage sensor FS-CVT01, temperature sensor FS-CTM01 and altitude sensor FS-CAT01 on all my models together with FS-IA6B and FS-IA10B receivers. 

I previously mentioned these issues in pr and issue nos #3515 and #3583

pfeerick, 2023-07-26:
Part of the problem with the FS-CAT01  is the iBUS processing is shared for both AFHDS 2A (NV14) and AFHDS3 (EL18) RF systems, and there seems to be some variance between how the two are handled. It was adjusted for the EL18, and verified against Flysky PL18 with stock firmware, but it cannonballed MPM and NV14 due to the AFHDS 2A being different. 

i.e. on the PL18, it reports altitude as being roughly -138M for me... With 2.9.0-RC1, I get a similar reading on the EL18, and would need to use offset (or auto offset) to bring it back to 0 if I want to correct for relative ground level. If I switch over to NV14 with the same sensor, it also shows -138-139 after a minute or two. It's only when I get to the TX16S w/ MPM that Alt goes crazy... -1005m, Tmp2 is weird, Pressure seems to be updating way too fast and wrong. The aforementioned PR seemed to fix this for the MPM, without affecting the other radios. Just had an issue outstanding for the FRM303 that if it still exists will need to be followed up separately.

richardclli, 2023-07-27:
The altitude is now calculated via a table lookup with pressure measured and the definition is the altitude above the sea level.

> Offset seems high, but can easily be compensated in "alt" telemetry settings or better via "Auto Offset" (both tested).
> A simple altitude test ( 1 floor) show's a 3.0 - 3.5m variation. Looks OK to me.

CharlesGauci, 2023-08-18:
Testing version 2.9 RC2 Providence with FS-CAT01 on NV14 using FS-IA6B/FS-IA10B receivers, and EL18 using FTR10 receiver.
NV14 : Seems to be working fine on Alt and Vspd vario, I will try it in the field next week
EL18 : Alt keeps constantly changing values, I tried using Offset and Auto Offset but amounts keep on changing and do not seem to be responding to actual altitude changes. Vspd is still not included in the telemetry screen, I tried to create a copy of Alt, renamed it to Vspd and changed it to m/s, but it did not work.

Al24650, 2025-01-06:
Hello, regarding this altitude error problem, I don't think it is related to the EdgeTX version only. Indeed, I have the TX16S radio control with EdgeTx 2.10.3 and I have two FS CAT01 sensors that work correctly with FS IAB6 receivers. I recently bought two other FS CAT01 sensors but they don't work on this same system. I don't understand, there seem to be several versions of this same sensor. I contacted FlySky to get answers on this subject. If I get answers I will publish them
---------
Bonjour, par rapport à ce problème d'erreur de l'altitude, je ne crois pas qu'elle soit liée à la version d'EdgeTX uniquement. En effet, j'ai la radiocommande  TX16S avec EdgeTx 2.10.3 et j'ai deux capteurs FS CAT01 qui fonctionnent correctement avec des récepteurs FS IAB6. J'ai recemment acheté deux autres capteurs FS CAT01 mais ils ne fonctionnent pas sur ce même système. Je ne comprends pas, il semble y avoir plusieurs versions de ce même capteur. J'ai contacté FlySky pour obtenir des réponses à ce sujet. Si j'obtiens des réponses je les publierai

dand3, 2025-01-06:
There are several "fake" version on the market. I have purchased at least 15# of them on Ali**ss (various sellers) and about 3 or 4 pces did not work or report erroneous value. I Just "bin" them (poubelle). You can try to make a claim eventually.. and buy from trusted source. I do not know of other reasons.

pfeerick, 2026-11-12:
I finally went through both telemetry pipelines (NV14 internal module and MPM) to see if/why the FS-CAT01 might behave differently. All decoding happens in EdgeTX, so the MPM is not to blame here. There was a bug introduced in 2.8.3 that I think was fixed in 2.10 onwards by #3582, which the 2.11/2.12 reports seem to confirm. 

**The non-zero starting altitude is by design.**
- Since 2.8.3, EdgeTX uses FlySky's own conversion table with a fixed sea-level reference (101320 Pa). It reports pressure altitude, so it won't start at 0 and will drift a little with the weather.
- Use **Auto Offset**, or reset the Alt sensor with a special function, to get height above takeoff.

**Why the NV14 "just worked" (sort of!): put simply, it doesn't always use the same code
- I tested a diagnostic build on an NV14 with RF module firmware 1.0.15. Firmware older than 1.1.2 sends pressure already converted to a 2-byte value, which EdgeTX decodes in a separate path (`flysky_nv14.cpp`).
- So the NV14 and MPM results were never a like-for-like comparison of the same decoder.
- NV14s with RF firmware 1.1.2 or newer go through the shared iBUS decoder that MPM uses. I couldn't test that combination.
- This also explains the missing VSpd (although that isn't the focus of this issue, so off-topic, so will be tracked via #3515) - the CAT-01 only sends pressure, so VSpd isn't something the sensor measures. The older NV14 decoder calculates it from successive altitude readings. The shared iBUS decoder used by MPM, the EL18 (AFHDS3) and newer NV14 RF firmware has no equivalent, so those setups don't get VSpd.

closed 2026-11-12 by pfeerick