EdgeTX/edgetx #3849: companion 2.9 RC1 read/write radio settings messes up throttle stick cal

ror1948, 2023-07-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

read and modify settings , run sim and then write back to tx16s caused this
tx16s has ver 2.10.0 nightly b0e64c8d 2023-7-11 2:46:57

throttle(maybe more) was way out of cal
recal fixed it
volume and backlight in sys global functions lost the previously assigned LS and RS control
I see volume and backlight settings are wrong in sim

### Expected Behavior

nothing messed up by doing a read and write in companion

### Steps To Reproduce

run companion 2.9
read radio settings
run sum on a model
close sim
same settings to radio

### Version

2.9.0

### Transmitter

Radiomaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### OS Version

win10 64bit

elecpower, 2023-07-24:
Hi @ror1948 normally it is strongly advised not to mix versions between Companion and radio firmware due to possible changes to settings introduced in newer versions.
With the two versions mentioned there are considerable differences resulting from the ADC refactor which broke settings for calibration as well as others. The latest Companion build is still awaiting fixes to be tested and merged before it will be back in sync with the firmware.
Therefore I will close this issue as a known incompatibility that will be fixed in due course.

closed 2023-07-24 by elecpower

pfeerick, 2023-07-25:
I would just like to add that using an older version of Companion with a newer version of the firmware is almost always guaranteed to break *something*.

ror1948, 2023-07-25:
but it is the latest version of companion available.
So you should stop using companion on nightly builds?

pfeerick, 2023-07-25:
It is the latest *release* version of Companion. You instead need a test/dev version of companion if you are going to use it in conjunction with a nightly firmware. But as Neil said, there are also still fixes pending to bring Companion in to line with changes made to the firmware

---

For the latest version testing version of Companion, from the link below for your preferred operating system, pick the most recent successful build (will have a green checkmark next to it). On the resulting page, the link to download the installer package is under the Artifacts heading near the bottom.

Companion (main branch): 
- Windows =>  https://github.com/EdgeTX/edgetx/actions/workflows/win_cpn-64.yml?query=branch%3Amain
- Mac => https://github.com/EdgeTX/edgetx/actions/workflows/macosx_cpn.yml?query=branch%3Amain
- Linux => https://github.com/EdgeTX/edgetx/actions/workflows/linux_cpn.yml?query=branch%3Amain