EdgeTX/edgetx #3851: Jumper T-Lite - Not able to enter LUA configuration mode (MDL doesn't do what it is supposed to). LUA "developer" says it's an EdgeTX or hardware issue

Kikerumbo, 2023-07-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Greetings,

I'm using the Telemetry Widget LUA script ( https://github.com/iNavFlight/OpenTX-Telemetry-Widget ) and I've been trying to configure the alarms (I use Li-Ion and the warnings are set too high) and It's not possible to enter the configuration with a Jumper T-Lite with EdgeTX.

The guide states that you should long press the "MENU" button to enter the configuration menu, but none of the buttons on the radio get me into it.

At the Github of the LUA ( https://github.com/iNavFlight/OpenTX-Telemetry-Widget/issues/103 ), they told me it's an EdgeTX or hardware issue but since the button works fine in other menus of the radio... here I am.

Here's the answer they gave me:

> Whatever, the code uses the firmware's `EVT_VIRTUAL_MENU_LONG` for the menu. If this is different in the simulator and the hardware, then it is an EdgeTX or vendor issue. Please contact them for help. The maintainer does not have access to the hardware.

Here's a video of the issue: https://youtube.com/shorts/nW9xRd1KelA?feature=share

Thanks.

### Expected Behavior

I press MDL (lower right button, as I was told by the maintainer of the LUA) and it does not enter the configuration menu of the LUA.

Any other button doesn't get me into the configuration menu either.

### Steps To Reproduce

1) Long press "DOWN" button to enter LUA screen.
2) Long press "MDL" button and nothing happens.

### Version

2.8.3

### Transmitter

Jumper T-Lite

pfeerick, 2023-07-27:
I'm unable to reproduce this sorry. I downloaded the [current release version](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/releases/tag/v2.2.1) of the iNav Lua and flashed my TLite back to v2.8.3. A long press of MDL worked to get into the iNav Telemetry Lua settings... 

https://youtu.be/tzXOnG3TeFA

However, it doesn't really look like a hardware issue either, as the button was responding to both long and short presses when you tried it outside of the script. Maybe try re-installing the Lua just in case something got corrupted. I also gave the event lua that was posted at iNavFlight/OpenTX-Telemetry-Widget#106 a try, and it correctly indicated that the MENU LONG event was fired on a long press of MDL. 

[image]

Kikerumbo, 2023-07-27:
> I'm unable to reproduce this sorry. I downloaded the [current release version](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/releases/tag/v2.2.1) of the iNav Lua and flashed my TLite back to v2.8.3. A long press of MDL worked to get into the iNav Telemetry Lua settings...
> 
> https://youtu.be/tzXOnG3TeFA
> 
> However, it doesn't really look like a hardware issue either, as the button was responding to both long and short presses when you tried it outside of the script. Maybe try again, but long press a bit longer... now that I look back at my own video, it seems a long press is a bit longer than I thought it was! Or try re-installing the Lua just in case something got corrupted. I also gave the event lua that was posted at [iNavFlight/OpenTX-Telemetry-Widget#106](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/discussions/106) a try, and it correctly indicated that the MENU LONG event was fired on a long press of MDL.
> 
> [image]

Thanks for taking the time. I tried reinstalling and updating the radio, same issue no matter how much time I press the button.

Since I can't follow the mwptools install guide (for a non-tech savvy, it's missing steps), I'll just use the "GPSx9L" script.

Thanks again.

closed 2023-12-24 by pfeerick