EdgeTX/edgetx #3853: [bug] #3745 fixes stick jitter , but wrongly reverses sticks for xlite

nerdCopter, 2023-07-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

**xLite (original v1)** left/right sticks are reversed, as well as stick defection is reversed in `main`.

i've gone through some commits to narrow down source of problems, using:
```
cmake -Wno-dev -DPCB=XLITE  -DDEFAULT_MODE=2 -DGVARS=NO -DPPM_UNIT=US -DCMAKE_BUILD_TYPE=Release ../     ## GVARS=YES too big for linking
make -C arm-none-eabi -j`nproc` firmware
```

(b6dd3a0ef8013ba93dfdc404debac2ea409d1b5f) works with stick re-calibration.
(2b100e0eb540a20cc6d357ec746562595086fdae) is broken with stick jitter (video #1).
all SHA in-between the above two fail due to GVAR breakage.
https://photos.app.goo.gl/myiW6qNEaUWzu5gP8

(2b100e0eb540a20cc6d357ec746562595086fdae) + cherrypick of fix: MIXSRC Heli & GVar (6617a5da6417b746d01f9980d727c72605c8110d) is broken with stick jitter (same as video #1) -- therefore, this is the commit that causes jitter.

PR #3745 fixes jitter, but wrongly reverses sticks and output. (video # 2)
https://photos.app.goo.gl/AReibsWHDGPHf8Gb9 (pardon mis-rotated video)

2.9.0-RC (`51b4341`) works

### Expected Behavior

sticks & stick-direction not reversed

### Steps To Reproduce

n/a

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X-Lite / S / Pro

### Operating System (OS)

Linux

raphaelcoeffic, 2023-07-27:
> GVARS=YES too big for linking

Are you compiling under MSYS2? (Windows?) Using regular build (has `GVARS=YES`) works for me as usual. It has still 13KB free head room in FLASH).

> wrongly reverses sticks and output

We need to be sure about one thing to properly fix this:
- X-lite v1 gimbal output a analog voltage (Hall gimbals with analog output, as opposed to the PWM signal used by X-lite Pro)

Is that correct?

raphaelcoeffic, 2023-07-27:
I created #3857 to try different settings. Would be nice if you could test the builds from the "Checks" tab and report there.

nerdCopter, 2023-07-27:
Ubuntu 22.04 and same in the past with 20.04.

M12 Hall sensor, PWM according to this: https://oscarliang.com/x-lite-latency-testing/

Will test #3863 as soon as i can.

pfeerick, 2023-07-28:
If you had DEBUG=Y enabled at some point, that would probably explain it, as I think DEBUG and GVAR was broken for a while. It's always good go clear your build folder and start again every once in a while, so that cached CMake config values don't ruin your day.

PR #3857 (2023-07-28): fix(x-lite): ADC channel definition
Fixes #3853

Summary of changes:
- make definitions consistent between `ADC_GPIO_PIN_STICK_XX` and `ADC_CHANNEL_STICK_XX`
- fixed `STICK_PWM_CHANNEL_XX` definitions
closed 2023-07-28 by pfeerick