EdgeTX/edgetx #3858: TX16s with 2.10.0 german selfbuild crashes in emergency screen

3DRacer, 2023-07-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

With a german build from master version 2.10.0 created in the browser the system crashes when you try to select the POTIS screen under SYS / Hardware

### Expected Behavior

The POTIS screen should be able to be selected, as it is in the English build.

### Steps To Reproduce

1. create a build with 
cmake -Wno-dev -DPCB=X10 -DPCBREV=TX16S -DDEFAULT_MODE=3 -DGVARS=YES -DPPM_UNIT=US  -DFWDRIVE=NO -DINTERNAL_GPS=NO -DTRANSLATIONS=DE -DCMAKE_BUILD_TYPE=Release ../
2. copy generated *.bin file to SD-Card and flash radio.
3. go to SYS / Hardware / Potis and select  them.
4. radio crashes

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Operating System (OS)

Windows

koaxheli, 2023-07-27:
There are other bugs in v2.10.0.

1. crash when opening POTIS because the sliders are now in the potifeld.
2. invert gas does not work either.
3. timers that are controlled with gas invert also run upside down.

Version
2.10.0 Selfmade German

Transmitter
Radiomaster TX16S / TX16SMK2

raphaelcoeffic, 2023-07-27:
> crash when opening POTIS because the sliders are now in the pot field

This might be related to your existing config. Can you please upload your `radio.yaml` file with which you can reproduce the issue?

> invert gas does not work either.
> timers that are controlled with gas invert also run upside down.

These are obviously different problems, which deserve to have an own issue. You can put the 2 things into the same issue, as they are probably related.

raphaelcoeffic, 2023-07-27:
> `-DFWDRIVE=NO -DINTERNAL_GPS=NO`

Note: these are the defaults, so you don't need to specify them ;-)

koaxheli, 2023-07-27:
My radio.yml is the following.

[main.zip](https://github.com/EdgeTX/edgetx/files/12184789/main.zip)

koaxheli, 2023-07-27:
Sorry My radio.yml is the following.
[radio.zip](https://github.com/EdgeTX/edgetx/files/12184850/radio.zip)

3DRacer, 2023-07-27:
I confirm fix: pot types strings #3863 solved that issue. 
Thanks

PR #3863 (2023-07-28): fix: pot types strings
Fixes #3858
commit 9f12025d (2023-07-28): fix: pot types strings (#3863)

Fixes #3858
closed 2023-07-28 by pfeerick