EdgeTX/edgetx #3872: Set Trim to Sub trim issues when using scroll wheel.

TB2185, 2023-07-28:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I'm always hitting the scroll wheel getting into the outputs tab and then it moves all my trims into sub trims. i mainly fly fuel planes this makes me have to reset up my idle settings. 

### Describe the solution you'd like

Moving the Set Trims to Sub Trim Tab down to bottom of the outputs page. 

### Describe alternatives you've considered

1. Changing the location of the Set Trims to Sub Trim Tab. On the Bottom of the outputs page. or?

pfeerick, 2023-07-29:
I mentioned this just now on discord, but will add it again here since it's relevant... (as well as expand on it a bit more)

On OpenTX, that button was originally at the bottom, and went missing for a while in UI transition from OTX to ETX. When it got re-added, it was then proposed that it be put at the top of the list, as most people didn't even realise the option was there. Plus it saved scrolling for it. So it's already been at the bottom, and deemed not suitable. 

So would instead a YES/NO confirmation of the change before it's made be better?

TB2185, 2023-07-29:
Yes a confirmation would just be fine.  I will as told the Outputs tap has a option to be disabled in 2.9 so that alone will help but the configuration would be best case scenario.

3djc, 2023-07-29:
Otx had a long enter required to activate. I find the top position to be a very bad idea, as it is a very rarely used function

TB2185, 2023-07-29:
> Otx had a long enter required to activate. I find the top position to be a very bad idea, as it is a very rarely used function

Even a Long press of the scroll wheel would be better. Like a few seconds press of the scroll wheel.

PR #5374 (2024-08-05): feat(color): add confirmation to 'Add all Trims to subtrims' button
Fixes #3872 

[image]
closed 2024-08-05 by pfeerick