EdgeTX/edgetx #3874: Global Variable use in Special Function Override channel

qsiguy, 2023-07-29:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

In the special functions when creating an "Override" value to a channel the only option is to enter a numeric value.  I would like to be able to specify a global variable to be the override value.

### Describe the solution you'd like

I would like a "GV" button added next to the "Value" field in Special Functions when Override is selected so you can select GV1, GV2, GV3, etc. just like you can do in the Inputs and Mixes next to Weight, Offset, and Curve.

### Describe alternatives you've considered

No suitable automated alternatives have been found.  At this time to perform my function I first capture real time values during flight using the "Adjust" special function which moves the desired values to several global variables.  I then later land the aircraft and manually move those captured values to the "Override" special functions to implement the desired override.

pfeerick, 2023-07-29:
You can have conditional replace mix lines, that mimic the functionality of an override... would that work?

i.e. the below uses a mix replace line to override the throttle with GV1 if SA is down.

[image]
[image]

qsiguy, 2023-07-29:
Yes, that should do the job, good idea.  Still seems like a good feature to add to the special functions... ;)

qsiguy, 2023-07-30:
After further review, while that method will work, if there are many other mixes already setup for the model like I normally do, it would be preferred to activate the override function to ensure no other mixes conflict with the activate of this logic.  It will work in the mixer but it will take more complex logic to force this particular function to take priority over all other mixes if they exist. On the model I am testing this on if I put it in the inputs, there are already 6 other "rates" setup there, and in the mixer there are 4 other mixes.  That is why I was using a special function for this in the first place.

It would seem to be a fairly simple addition to the special function since the code already does it in the inputs and mixer.  If it doesn't happen, I'll make it work with a hybrid of your idea above.  Thanks for taking the time to provide an option!