EdgeTX/edgetx #3882: Splash screen timing overidden by stick/switch position warnings

Kevltan, 2023-07-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Splash screen timing is overidden by stick & switch position warning announcement and reverts to current model screen.

### Expected Behavior

When splash screen set time is finished, stick and switch position warnings are announced with current model screen displayed.

### Steps To Reproduce

1. Power up.
2. Splash screen comes on.
3. If any stick or switch is out of preset position, reverts to model screen after approx 2 secs and warning is announced, even if max splash screen time is selected.

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

2.10.0 - nightly

pfeerick, 2023-07-30:
I thought this someone would mention this as soon as the configurable delay option got noticed 😆

Kevltan, 2023-08-01:
Works perfect now - as expected to work👍👌😊

pfeerick, 2023-08-03:
As long as you don't DFU flash 2.10 / nightlies / PRs (i.e. use bootloader to flash), you should still be able to bootloader flash - at worst, with a hard power cycle if the MCU is locked up.

Kevltan, 2023-08-08:
Now works fine after latest update on both my TX16s and my X12S radios. 08-08-23 #7382

PR #3887 (2023-09-07): feat(radio): Delay warnings until after splash screen finished
Fixes #3882 

The startup code needs some attention and cleanup; but this should work for now.

Tested on TX16S and X7S.
closed 2023-09-07 by pfeerick