EdgeTX/edgetx #3883: ELRSV3 Lua only works on 115k and not 400k

ror1948, 2023-07-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

lua script ELRSV3 will not display ELRS info from BetaFPV Micro Tx 1W when at 400k
works at 115K

### Expected Behavior

lua script ELRSV3 displays ELRS info 

### Steps To Reproduce

open lua script ELRSV3

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

2.10.0 nightly b0e64c8d
2023-07-11 at 2:46:57

ror1948, 2023-07-30:
it works at 5.25m !!

pfeerick, 2023-07-31:
Can't replicate this I'm sorry... using the current nightly and TX16S, ELRS3 Lua is working for me at 400k (which is good as I almost always use 400k) for HM24TX @ v3.0.0, HM915TX @ v3.3.0-RC1, BetaFPV 2.4 Nano @ v3.1.2

What firmware version is your module running?

ror1948, 2023-07-31:
V3.3.0 released

it works ok at 5.25 baud

pfeerick, 2023-07-31:
It would be interesting to know if flashing that module back to 3.2 fixes it... as 3.3 includes a change to the autobaud calculation used in the ELRS firmware to connect with the handset... maybe you have a edge case where it doesn't work? It would at least rule that out as a possibility. 

https://github.com/ExpressLRS/ExpressLRS/pull/2283

Quas7, 2023-09-03:
additionally on TX12MK2 the ELRSv3 lua script does load forever with 3.75Mbaud but it works with higher 5.25Mbaud and lower 1.87Mbaud and also 400kbaud.
Note that 3.75Mbaud worked for me for edgeTX 2.8.x although I had occasionally telemetry and sensor losses that might hint an unreliable serial link.
Sticking to 1.87Mbaud now to be on the safer side.

raphaelcoeffic, 2023-09-03:
> Sticking to 1.87Mbaud now to be on the safer side.

It might be related to some timing that are now slightly different either on our or on ELRS side, thus making it a bit wonky.

Most of the time, achievable speed on half-duplex external link is .... well.... based on a combination of available hardware options (rx/tx DMA), interrupt load on that particular radio, etc. Not something we can change that easily.

Quas7, 2023-09-03:
Just quickly checked this on my Zorro as well. 3.75Mbaud does not load there either now but hardware is likely very similar to TX12 in this regard.

As @raphaelcoeffic already said, I also believe this is strongly related to the hardware/software combination with timing/oszillators/interrupts etc. 

Maybe it is a good idea to just give somewhere a prominent hint that ELRS lua loading issues can be related to module baud rates and close the issue with this?

3djc, 2023-09-03:
Yeah, because my RM nano module works on Zorro at all speed above 921k (didn't bother testing bellow), so it is really a combination issue sometimes