EdgeTX/edgetx #3888: Unable to select the ASSAN protocol in a 4in1 Radiomaster Tx Module

jims1234, 2023-07-31:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Other (Please specify below)

### Current Behavior

I have an EXTERNAL Radiomaster 4in1 module in a new TX16S MK2 with Internal ELRS. when I attempt to bind Frsky X D16 and D8 modules it works fine.. and When I went to bind an ASSAN Rx the very 1st time it worked fine that one time also.. but ever since I never get the ASSAN protocol option to appear in the alphabetic list,  mine always starts at "Bayang" 

I have tried re-flashing this EXT 4in1 module two times it was sucessful with both the original v1.3.3.20.bin it came with and v1.3.3.7.bin. however the symptom remains identical.  It worked ONCE and only once with my 1st ASSAN Rx but I havent been able to program any others since.. I have EspressLRS Rx's on order, But I also have many slope gliders and parkflyers with ASSAN Rx's I'd like to use with thie new Radio.. Please help.. I also posted this issiue as number #820 in the multi thread and Pascal suggest I ask for your help since he blelives it's a EdgeTx issue. 
   

### Expected Behavior

I'd like to see the ASSAN protocol option appear so I can Bind my Rx's 

### Steps To Reproduce

1.) select a new or existing model
2.) select Model Settings
3.) scroll to select Internal RF and trun off the internal ELRS module
4 .) scroll to select External RF and select MULTI/ RF Protocol "Std"
5. ) select "ASSAN" if you can.. ( but I can't) see this option..
6. Then If I could select it,  I would select "Bind"  but that option never appears under any RF Protocol   

### Version

2.8.4

### Transmitter

Radiomaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### OS Version

Win11 Pro

### Anything else?

later if desired/ needed

pfeerick, 2023-08-01:
I know this isn't the answer you want, but it works for me with TX16SMK2 ELRS v2.8.4 and RM 4IN1 MPM 1.3.3.20 🤭 

If you turn the external MPM off and restart the radio, do you get the "MPM: Scanning protocols" progress bar? Is the Assan protocol in the list once that scan has been done? If you get that progress bar *AND* no Assan in the list, we have problems... maybe the firmware on the MPM is wrong and it isn't reporting that the MPM supports Assan, or something else strange is going in. 

There is a bug that will be fixed in 2.8.5 that affects when you have a radio with both internal AND external MPM, but that doesn't apply to when you have internal or external only.

jims1234, 2023-08-01:
Wow, I think you've nailed the work around.. The "No ASSAN" option problem is definately repeatable in the current and prior MULTI Firmware releases.. and simply switching MULTI mode OFF and ON will not fixit.. 

However if I do as you asked.. which was to turn the EXT MULTI module OFF 1st, then POWER cycle the Radio Switch OFF, and back ON.. as you re-enable MULTI on that Ext module it then does that MULTI module scanning progress bar and when that completes in a few seconds .. then wala.. the ASSAN protocol re-appears on the top of the list.. Its also quite easy for it lt lock up again in the "No Assan available" mode but since the " set it off then Power OFF /On cycling fixes it,  I don't believe this is a significant show stopper.. I just hope I don't need to re-bind all my D16/D8 FrSky X planes now as a result of switching the firmware as I did.. 
Thanks a lot .. I had no idea that EdgeTx was Scanning MULTI modules for some reason. Thank you very much for a temp work around for this issue.

pfeerick, 2023-08-01:
Ah, I think I know what your problem is... it's *switching* models that loses the full list. If you power cycle the radio (no need to set the module off, power off, power on, set module off) ... just change model and power cycle... it should work fine. That is the other half of the fix that will be in 2.8.5... which will be out _"any day now"_...

jims1234, 2023-08-06:
Peter , 

thanks the new and improved/ abbreviated fix you suggest below woks fine.  Thanks. jims

Ref issue #3888

closed 2023-10-01 by pfeerick