EdgeTX/edgetx #3889: FrSky X-Lite is missing D8 protocol internal TX

BoxedBrain, 2023-07-31:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

xlite-edgetx-v2.8.4.bin is missing D8 protocol for the internal TX module.

### Expected Behavior

D8 should work as in prev. installed open-tx 2.3.15

### Steps To Reproduce

Flash latest available Firmware. D8 is missing.

### Version

2.8.4

### Transmitter

FrSky X-Lite / S / Pro

### Operating System (OS)

Windows

### OS Version

Windows 11

pfeerick, 2023-08-01:
What internal RF does that radio have... ISRM or XJT? With ISRM, it is ACCESS or D16 only... with XJT, you should get D8/D16/LR12. It seems like for the X-Lite it is XJT, but X-Lite S and X-Lite Pro both have ISRM.  You should be able to go to RADIO/SYS, HARDWARE, and change the Internal RF from ISRM to XJT.

BoxedBrain, 2023-08-01:
Thanks @pfeerick that solved it. Was this my fault, or is the build for the X-Lite (XJT) wrong?

raphaelcoeffic, 2023-08-02:
`xlite` target defaults to `XJT`:
```cmake
  set(DEFAULT_INTERNAL_MODULE XJT_PXX1 CACHE STRING "Default internal module")
```

So I guess something else happened. Maybe a `xlites` firmware was loaded and changed that, or maybe something else.

pfeerick, 2023-08-02:
A clean build of xlite `simu` also defaults to XJT. However, there has been the odd report that the internal RF is wrong, especially when conversion is involved, so there is that also. But at least this is working for you now, so I think we can chalk this up to being resolved for now.

closed 2023-08-02 by pfeerick