EdgeTX/edgetx #3891: Cannot run the latest firmware on my Horus X12S

Kevltan, 2023-08-01:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Loaded latest firmware and locks up on restart. Have to disconnect battery and load older firmware from STM32 cube programmer to re-initiate and get back to working condition. I have tried loading bootloader for the latest but this does not work either. I have tried a few times now and always with the same result. I am re-loading firmware from the V2.9.0  (ebef94d9) 2023-03-27 and this works fine.

### Expected Behavior

 Latest firmware to run after being installed from bootloader.

### Steps To Reproduce

1. load latest firmware on to disk
2. power off and into bootloader
3. load latest firmware
4. Exit and TX to start
5. No blue start light
6. Unable to connect to USB until internal battery disconnected
7. Then able to load old firmware from STM32 and TX restarts as before

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X12

### Anything else?

V2.9.0  (ebef94d9) 2023-03-27 runs fine

raphaelcoeffic, 2023-08-07:
Can you please re-try with the latest build from #3810 ?

Kevltan, 2023-08-07:
Yes will try later today👍

Kevltan, 2023-08-07:
This PR works for my X12s. 😊

Kevltan, 2023-08-07:
Can you please put the same edit in the firmware that allows the splash
screen to load, without being influenced by stick or switch position
notifications until the time delay has elapsed. As done by philmoz #3887
for the TX16S? 😊

pfeerick, 2023-08-07:
Wait for #3810 and #3887 to get merged, then you'll have that functionally
in the following nightly build.

closed 2023-08-08 by pfeerick