EdgeTX/edgetx #3901: Fix proper focus

JimB40, 2023-08-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

[image]

Focus on variuos UI interactive elements is now tiny frame.
@philmoz have you changed anything in global LVGL UI elements style definitions? 

### Expected Behavior

[image]

### Steps To Reproduce

Flash 2.10 nightly

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S / TX16SMK2

philmoz, 2023-08-04:
Focus was changed in the libopenui refactor PR to be more consistent (along with size, border and rounded corners).

Controls that can be both active and focused used the border color to show focus. 
That meant there were two types of focus - border or fill.

philmoz, 2023-08-04:
Keep in mind one of the main reasons for the libopenui refactor is to set up a better framework for 3.0.

JimB40, 2023-08-04:
Bit confused Phil.
There are style-sheets for UI elements prepared that include NORMAL, FOCUSED, ACTIVE and ACTIVE-FOCUSED states.
in case of above mentioned elements. https://github.com/EdgeTX/edgetx/issues/3225

As principal UX rule consistency can't override visibility.
Focus as inversed field is well defined and visible while tiny frame is not.
If there are tougher cases like Lists items or groups that uses tiny frame we shoud find a way to change those cases and not to equalize everything to worse case.

AFAIU libopenui now is under our full control so I can't see the point.

If there are implementation (because of how code is written)  I'm ready to discuss it to find best solution but not to sacrifice principals.

pfeerick, 2023-08-04:
Before we could even contemplate looking at the stylesheets, the code needed to be unified so things weren't been formatted and modified on every other page. Now that that has been done, starting to implement the stylesheet and progressing on to the 3.0 UI should be easier. 

Your job @JimB40 is to be starting to pull out these details, and pointing to the stylesheet to show how you want it to look, and it'll either happen or you'll be told it's not feasible. ;)

JimB40, 2023-08-04:
Hmm... Never seen case in my UX/UI career that "code unification" made styling not feasible.
But then okay. Wait and see.

At least EDIT state was reclutant to "refactor" or whatever and is no lookin like this
[image]

pfeerick, 2023-08-04:
> Hmm... Never seen case in my UX/UI career that "code unification" made styling not feasible. 

🤣 And who said that, other than you? 😛 You'll be told it's not feasible because of resource constraints like memory, processing time, ease of implementation, etc. But remember, baby steps... can't get to the style stuff, until the foundations are ready to actually do it. ;)

> At least EDIT state was reclutant to "refactor" or whatever and is no lookin like this

It depends on what you call "EDIT state" - it helps if you're just going to pull screenshots out to highlight a specific area of interest - do you mean the entire dialog? The input field for "input name" seems to be in focus, but not being edited? If you mean the latter, then I thought that was almost exactly what it looks like...

[image]

If you mean the entire edit dialog, no, the refactor was **_not_** related to that at all. For that, you need to work on trying to get https://github.com/EdgeTX/edgetx/pull/2939 how you want it. I stress *try* ... you will not necessarily get it exactly how you want it.

JimB40, 2023-08-04:
EDITED state is how UI input elements looks when user does editing. 
It's desctibed in UI element styling sheet (if UI element has EDITED state) like #3223
So FOCUSED in now tiny frame around while EDITED remained green inversion after "refactor" or whater it was done.

philmoz, 2023-08-04:
Border focus indicator is not something new I have added. I've just made it more consistent.

I'm not sure why this is causing so much angst.

Examples from 2.8:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

JimB40, 2023-08-04:
Because inversion is far better visible in direct sun, and that is where I often use radio.

pfeerick, 2023-08-04:
OK, so you meant "Text Input field EDIT state" ... gotcha. 
[image]

So, if we focus on just the one input field type... what you are calling "Text Input field", it currently looks mostly like the example ENABLED state. 

But when it's in the edited state it's filled with EDIT color, border is EDIT color and text color is PRIMARY2. And when it's FOCUSED, rather than it just being an outline, you want the entire control filled with the FOCUS color, border also FOCUS color and TEXT PRIMARY2? Correct?

JimB40, 2023-08-04:
Yep.
Have idea that would be good to have half and hour in next dev-hour. So I can explain EdgeTX UI System. 
So if I write later just a name from UI world we both know meaning. So I don't have to explain what UI Atom is :)
[image]

pfeerick, 2023-08-04:
Or perhaps you could learn the names of the UI controls that we use all the time, so we don't need to know two lots of terminology 🤪 But yes, that would be a good idea for the next one, as things are getting to a state where this can start happening. 

It's more... keep in mind... how would you explain it if you were face to face... wouldn't you point at what you are talking about? :wink: After all, pulling up a screenshot of the whole screen, and then babbling about EDIT state when in fact you're talking about a particular control isn't quite conveying the message you intended.

JimB40, 2023-08-04:
Somehow to communicate well in ETX project  I've learned what is behind "commit" or "rebase" to understand, not asking to use "my language". Learned also how to use GitHub even it was not on my "fun to do" list. 
Ussualy worked with front-end devs who knew terminology, so it wasn't my task to check if property for padding is actually named "padding" in CSS or whatever they used. 

If this is going to be boring-half-an-hour-I-dont-need scratching idea now.
After 50 "price" of hour of my life skyrocketed. So I try not make any waste. :)

PS.
Have to think over attitude, maybe @philmoz is right I'm causing too much. This is RC hobby project, not commercial one for big brand.
Cheers :)

pfeerick, 2023-08-04:
> I'm causing too much

That I can't agree with at all... as much as I would like to get hold of you and shake you at times (still sure you want that face-to-face meet? 🤣 ) ... it's just about trying to find that common terminology that we all understand. As you say, this is hobby/open source, not a commercial project, so gotta be more flexible ;)

pfeerick, 2023-12-12:
I'mma gonna stomp on this issue for now. Bring it up again as part of the 3.0 UI if still want it changed. IMO, this made things more consistent though...

i.e. **Before**
https://github.com/EdgeTX/edgetx/assets/5500713/1e20074e-6a5f-4462-86bd-680708e7c895

**After**
https://github.com/EdgeTX/edgetx/assets/5500713/732e5e1f-f5d9-429e-a980-3c4a785fae9e

closed 2023-12-12 by pfeerick