EdgeTX/edgetx #3906: low cost GPS not supported

sdellava, 2023-08-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Currently, adding a GPS to a radio requires a specific configuration (NEMA, 9600 Bps).

Unfortunately, many small GPSs do not have an internal flash and if they are not preconfigured in the correct mode, EDGTX will not recognize and use them.

My suggestion is to add an audobaud configuration and GPS initialization step when selecting a model.

GPS inizialization example can be seen here: https://github.com/betaflight/betaflight/blob/master/src/main/io/gps.c (gpsInitData_t)

### Expected Behavior

GPS non recognized and initialized.

### Steps To Reproduce

Connect a BN180 GPS or Flywoo GPS 2 and you will see it do not work

### Version

2.9.0

### Transmitter

Jumper T-Pro

### Operating System (OS)

Windows

sdellava, 2023-08-09:
Update: actually most of the low cost GPS are able to autoconfigure the port speed, but many of them are configured to use UBLOX protocol by default instead of NMEA. So It could be useful to add the UBLOX protocol to the GPS interface. In the BETAFLIGHT repoository there is the required code.

pfeerick, 2024-04-08:
Support for UBX GPSs has started with #4689

closed 2024-04-08 by pfeerick