EdgeTX/edgetx #3907: Special Functions list resets selected index after deletion [v2.85]

MartinHugh, 2023-08-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

On the Special Functions page

If I want to delete a block of entries one by one, I select one using the enter button, and select delete and press the enter button
The item is deleted and the user is returned to the position in the list that they were on before the deletion - which is all as expected.

However if I then try to scroll onto the next item in the block, I realise that the selected list index has been reset to the first element in the list.  

This means when I am deleting (in my case six entries from 35 to 40) I have to scroll all the way back from the first index into the 30's for each deletion I wish to do.

I understand that I could have possibly used the touch screen to reselect the list after deletion, but this should not be for the user to do.  It makes no sense anyway resetting the index to the start of the list. 

### Expected Behavior

Following deletion, the list should be reselected at the position which was just deleted.

### Steps To Reproduce

See current behaviour

### Version

2.9.0

### Transmitter

Radiomaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### OS Version

10

### Anything else?

The dropdown in the form does not allow me to select v2.85 which is the version I have the problem with.

philmoz, 2023-08-05:
This is fixed in 2.9.

raphaelcoeffic, 2023-08-08:
As this is fixed in `v2.9`, which is imminent, we won't do anything in `v2.8.x`.

closed 2023-08-08 by raphaelcoeffic